mit-han-lab/bevfusion #103: how is the difference of fusion at different position

AndyYuan96, 2022-08-17:
Hi, I just wonder whether your guys have ever do experiment about different fusion position, as I see your guys fusion before backbone, and bevfusion of Alibaba fusion after neck, so I just wonder the difference of fusion position.

kentang-mit, 2022-08-17:
We did not try fusion after neck in our experiments and this might be an interesting direction to explore in the future. Fusing before the BEV backbone may lead to a simpler design for the fusion module: it is unnecessary to add additional operators (such as SE) to the fuser because there are a lot more conv layers after fusion.

AndyYuan96, 2022-08-17:
> We did not try fusion after neck in our experiments and this might be an interesting direction to explore in the future. Fusing before the BEV backbone may lead to a simpler design for the fusion module: it is unnecessary to add additional operators (such as SE in the PKU-Alibaba paper you mentioned) to the fuser because there are a lot more conv layers after fusion.

thanks for quickly reply.

closed 2022-08-17 by AndyYuan96