mit-han-lab/bevfusion #105: OptionalCUDAGuard

FuzhiYang, 2022-08-17:
Hi Haotian,

In the recent commit, "OptionalCUDAGuard" is added in mmdet3d/ops/bev_pool/src/bev_pool.cpp. I just wonder what is the purpose of this module? 
I built the code in the old version (without OptionalCUDAGuard). Will this cause some potential errors? 

Thanks,
Fuzhi

kentang-mit, 2022-08-17:
Hi Fuzhi,

This will not cause potential errors if you are not debugging the single BEV pooling op. `OptionalCUDAGuard` is related to the discussions in [this issue](https://github.com/mit-han-lab/bevfusion/issues/63). 

Best,
Haotian

FuzhiYang, 2022-08-18:
Got it. Thank you very much!

kentang-mit, 2022-08-18:
No problem. I'm closing this issue temporarily because the issue is resolved.

closed 2022-08-18 by kentang-mit