mit-han-lab/bevfusion #106: undefined symbol: _ZN2at5sliceERKNS_6TensorElN3c108optionalIlEES5_l

hmy-meng, 2022-08-18:
Dear Haotian,Thanks for your great work!
I am trying to evaluate the detection model on a single GPU machine by running the following in terminal:
torchpack dist-run -np 1 python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth --eval bbox
it gives me:

Traceback (most recent call last):
  File "tools/test.py", line 14, in <module>
    from mmdet3d.apis import single_gpu_test
  File "/home/ubuntu/workspace/bevfusion/mmdetection3d/mmdet3d/apis/__init__.py", line 2, in <module>
    from .train import *
  File "/home/ubuntu/workspace/bevfusion/mmdetection3d/mmdet3d/apis/train.py", line 15, in <module>
    from mmdet.core import DistEvalHook
  File "/home/ubuntu/workspace/bevfusion/mmdetection/mmdet/core/__init__.py", line 3, in <module>
    from .bbox import *  # noqa: F401, F403
  File "/home/ubuntu/workspace/bevfusion/mmdetection/mmdet/core/bbox/__init__.py", line 8, in <module>
    from .samplers import (BaseSampler, CombinedSampler,
  File "/home/ubuntu/workspace/bevfusion/mmdetection/mmdet/core/bbox/samplers/__init__.py", line 10, in <module>
    from .score_hlr_sampler import ScoreHLRSampler
  File "/home/ubuntu/workspace/bevfusion/mmdetection/mmdet/core/bbox/samplers/score_hlr_sampler.py", line 3, in <module>
    from mmcv.ops import nms_match
  File "/home/ubuntu/anaconda3/envs/bevf/lib/python3.8/site-packages/mmcv/ops/__init__.py", line 2, in <module>
    from .assign_score_withk import assign_score_withk
  File "/home/ubuntu/anaconda3/envs/bevf/lib/python3.8/site-packages/mmcv/ops/assign_score_withk.py", line 5, in <module>
    ext_module = ext_loader.load_ext(
  File "/home/ubuntu/anaconda3/envs/bevf/lib/python3.8/site-packages/mmcv/utils/ext_loader.py", line 13, in load_ext
    ext = importlib.import_module('mmcv.' + name)
  File "/home/ubuntu/anaconda3/envs/bevf/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: /home/ubuntu/anaconda3/envs/bevf/lib/python3.8/site-packages/mmcv/_ext.cpython-38-x86_64-linux-gnu.so: undefined symbol: _ZN2at5sliceERKNS_6TensorElN3c108optionalIlEES5_l
--------------------------------------------------------------------------
Primary job  terminated normally, but 1 process returned
a non-zero exit code. Per user-direction, the job has been aborted.
--------------------------------------------------------------------------
--------------------------------------------------------------------------
mpirun detected that one or more processes exited with non-zero status, thus causing
the job to be terminated. The first process to do so was:

  Process name: [[11928,1],0]
  Exit code:    1
--------------------------------------------------------------------------

I am not sure what could be the problem, could that be due to the version does not match, or something else?
A lot of thanks for your kindly reply .
The version is as follows:
Python==3.8
cuda==11.2
torch==1.10.0                  
torchpack==0.3.1                  
torchvision==0.11.1       
mmcv-full==1.4.0     
[nuscenes-dev-kit](https://github.com/nutonomy/nuscenes-devkit)==1.1.9               
mmdet==2.20.0                    
mmdet3d==1.0.0rc4               
mmsegmentation==0.27.0   
OpenMPI==4.0.4 and mpi4py = 3.0.3 
Pillow==8.4.0
[tqdm](https://github.com/tqdm/tqdm)==4.64.0

VeeranjaneyuluToka, 2022-08-18:
Hi, I have got exactly same error but it was due to some installation for sure. I have resolved by changing some versions but do not really remember what helped me, however you can find out below the installed version in my machine (rtx 3090 ti) and highlighted differences with bold text.

Python==3.8.0

cuda==11.3
torch==1.10.0
torchvision==0.11.1
Have used the below command to install pytorch
conda install pytorch==1.10.0 torchvision==0.13.1 torchaudio==0.10.0 cudatoolkit=11.3 -c pytorch -c conda-forge

torchpack==0.3.1
mmcv-full==1.4.0
[nuscenes-dev-kit](https://github.com/nutonomy/nuscenes-devkit)==1.1.9
mmdet==2.20.0
**mmdet3d==0.0.0**
**OpenMPI==4.0.2** and mpi4py = 3.0.3
Pillow==8.4.0
[tqdm](https://github.com/tqdm/tqdm)==4.64.0

kentang-mit, 2022-08-18:
Thanks @VeeranjaneyuluToka for the discussion! The library versions are indeed quite important for the current BEVFusion repo. I would suggest @hmy-meng to also have a closer look at the mmcv and mmdet versions and make sure your configurations match our recommendations.

hmy-meng, 2022-08-19:
@VeeranjaneyuluToka @kentang-mit   Thanks a lot for your answer, I solved it, using this method ：https://github.com/mit-han-lab/bevfusion/issues/13#top

kentang-mit, 2022-08-19:
Great to hear that. I'm temporarily closing this issue since it is resolved.

closed 2022-08-19 by kentang-mit