mit-han-lab/bevfusion #107: RuntimeError: CUDA error: no kernel image

junwenjiao, 2022-08-19:
when I run `torchpack dist-run -np 8 python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth --eval bbox`
It encounter a problem as mentioned below:
`RuntimeError: CUDA error: no kernel image is available for execution on the device
CUDA kernel errors might be asynchronously reported at some other API call,so the stacktrace below might be incorrect.
For debugging consider passing CUDA_LAUNCH_BLOCKING=1.
`
I think everything before the BEV pooling kernel (especially the sparse convolution backbone) runs .But I do not know how to solve it.

cheukcat, 2022-08-19:
RuntimeError: CUDA error: no kernel image is available for execution on the device
CUDA kernel errors might be asynchronously reported at some other API call,so the stacktrace below might be incorrect.
For debugging consider passing CUDA_LAUNCH_BLOCKING=1.
--------------------------------------------------------------------------
Primary job  terminated normally, but 1 process returned
a non-zero exit code. Per user-direction, the job has been aborted.
--------------------------------------------------------------------------
--------------------------------------------------------------------------
mpirun detected that one or more processes exited with non-zero status, thus causing
the job to be terminated. The first process to do so was:

  Process name: [[27088,1],1]
  Exit code:    1

I got this same

kentang-mit, 2022-08-19:
Do we have more information about the environment? E.g. type of GPU, version of CUDA, cuDNN and PyTorch? Also, please make sure that you install the correct versions of required libraries indicated in our README.

cheukcat, 2022-08-19:
> 

My CUDA version is 11.1, my PyTorch version is 1.9.0 and I'm pretty sure that my environment is aligned with cu111. The whole conda env is installed following README.
This problem occurs while I trying to inference with a pretrained BEV segmentation models using
`torchpack dist-run -np 8 python tools/test.py configs/nuscenes/seg/fusion-bev256d2-lss.yaml pretrained/bevfusion-seg.pth --eval map`
The error is actually happens after the QuickCumsumCuda.apply, and it goes like this
`  File "/cbh/bevfusion/mmdet3d/models/vtransforms/base.py", line 167, in bev_pool
    x = bev_pool(x, geom_feats, B, self.nx[2], self.nx[0], self.nx[1])
  File "/cbh/bevfusion/mmdet3d/ops/bev_pool/bev_pool.py", line 96, in bev_pool
    x = x.permute(0, 4, 1, 2, 3).contiguous()
RuntimeError: CUDA error: no kernel image is available for execution on the device
CUDA kernel errors might be asynchronously reported at some other API call,so the stacktrace below might be incorrect.
`

junwenjiao, 2022-08-19:
my CUDA version is 11.3.
GPU:GP107 [GeForce GTX 1050 Ti]
pytorch:1.10.0  
other required libraries are installed indicated in your README
but do not intall cudnn

junwenjiao, 2022-08-19:
and like issue #26.I  removing all -gencode=arch=compute_XX,code=sm_XX in the setup.py.

cheukcat, 2022-08-19:
> and like issue #26.I removing all -gencode=arch=compute_XX,code=sm_XX in the setup.py.

I also tried but it doesn't work

junwenjiao, 2022-08-19:
I also change "distributed"as I think I need to run single_gpu_test()

kentang-mit, 2022-08-19:
Hi folks, 

Would you mind also taking a look at [this issue](https://github.com/mit-han-lab/bevfusion/issues/101)? For @junwenjiao's environment, I think you probably want to add something like `compute_61` and `sm_61` (corresponding to GTX1050Ti) to the setup file and rerun the setup develop command. 

Besides I try to find some time setting up a Docker for the environment. But it takes some time.

Best,
Haotian

cheukcat, 2022-08-20:
> Hi folks,
>
> Would you mind also taking a look at [this issue](https://github.com/mit-han-lab/bevfusion/issues/101)? For @junwenjiao's environment, I think you probably want to add something like [...]

Hi Haotian, corresponding to GTX 1080ti, what should I add to the setup file?

kentang-mit, 2022-08-22:
Sorry for the late reply. For 1080Ti, it is also `compute_61` and `sm_61`.

junwenjiao, 2022-08-30:
thanks for your help.I try it but there is nothing happened.

cheukcat, 2022-09-01:
> 

My problem is solved by using compute_61 and sm_61, thx

kentang-mit, 2022-09-01:
Glad to hear that. I'm closing this issue because it is resolved.

closed 2022-09-01 by kentang-mit