mit-han-lab/bevfusion #109: Problem when evaluating

TheFatBlue, 2022-08-19:
Hi, when I run the command `torchpack dist-run -np 8 python tools/test.py configs/nuscenes/seg/fusion-bev256d2-lss.yaml pretrained/bevfusion-seg.pth --eval map`, I got the output below:

> 2022-08-19 06:50:26,094 - mmdet - WARNING - No pre-trained weights for SwinTransformer, training start from scratch
> load checkpoint from local path: pretrained/bevfusion-seg.pth
> [                                                  ] 0/6019, elapsed: 0s, ETA:~/anaconda3/envs/fusion/lib/python3.8/site-packages/nuscenes/map_expansion/map_api.py:1823: ShapelyDeprecationWarning: Iteration over multi-part geometries is deprecated and will be removed in Shapely 2.0. Use the `geoms` property to access the constituent parts of a multi-part geometry.
>   exteriors = [int_coords(poly.exterior.coords) for poly in polygons]
> ~/anaconda3/envs/fusion/lib/python3.8/site-packages/nuscenes/map_expansion/map_api.py:1824: ShapelyDeprecationWarning: Iteration over multi-part geometries is deprecated and will be removed in Shapely 2.0. Use the `geoms` property to access the constituent parts of a multi-part geometry.
>   interiors = [int_coords(pi.coords) for poly in polygons for pi in poly.interiors]
> [>>] 6024/6019, 8.7 task/s, elapsed: 695s, ETA:     0s

Then the process got stuck: there was no more output and the process didn't terminate. This went on for about an hour. During this, I run the command `nvidia-smi` to check the GPUs(eight Tesla M40) and found that the memory usage is approximately 8000MiB / 12000MiB and GPU-Util is 0% except the first is 100%. 
Is this normal? Could you please help me with the problem? thanks!

kentang-mit, 2022-08-19:
Thank you for the information. it seems that the evaluation is finished, but the results are not showing. I would recommend two possible ways to fix that:

1. Would you mind trying adding "-v" to the command to see whether there are some error messages?

```bash
torchpack dist-run -np 8 -v python tools/test.py configs/nuscenes/seg/fusion-bev256d2-lss.yaml pretrained/bevfusion-seg.pth --eval map
```

2. Would you mind trying using 1 GPU to perform the evaluation and see whether the same problem occurs again?

Best,
Haotian

TheFatBlue, 2022-08-20:
thank you for your suggestions! 
I tried the first one but got stuck after the output below:

load checkpoint from local path: pretrained/bevfusion-det.pth
[                                                  ] 0/6019, elapsed: 0s, ETA:/home/eecs/shz/anaconda3/envs/fusion/lib/python3.8/site-packages/torch/functional.py:445: UserWarning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (Triggered internally at  /opt/conda/conda-bld/pytorch_1634272128894/work/aten/src/ATen/native/TensorShape.cpp:2157.)
  return _VF.meshgrid(tensors, **kwargs)  # type: ignore[attr-defined]
/home/eecs/shz/anaconda3/envs/fusion/lib/python3.8/site-packages/torch/functional.py:445: UserWarning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (Triggered internally at  /opt/conda/conda-bld/pytorch_1634272128894/work/aten/src/ATen/native/TensorShape.cpp:2157.)
  return _VF.meshgrid(tensors, **kwargs)  # type: ignore[attr-defined]

for the second one, I encountered the same problem in this [issue](https://github.com/mit-han-lab/bevfusion/issues/41#issue-1283444106). now I'm trying to evaluate the camera-only model mentioned in that issue

TheFatBlue, 2022-08-21:
I run the evaluation command with single/fusion model on det and seg task separately. With single model, I can get the expected output, and with the fusion model, I got the error again:

Traceback (most recent call last):
  File "tools/test.py", line 230, in <module>
    main()
  File "tools/test.py", line 203, in main
    outputs = multi_gpu_test(model, data_loader, args.tmpdir, args.gpu_collect)
  File "/home/mmdetection/mmdet/apis/test.py", line 98, in multi_gpu_test
    result = model(return_loss=False, rescale=True, **data)
  File "/home/anaconda3/envs/fusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/anaconda3/envs/fusion/lib/python3.8/site-packages/torch/nn/parallel/distributed.py", line 886, in forward
    output = self.module(*inputs[0], **kwargs[0])
  File "/home/anaconda3/envs/fusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/anaconda3/envs/fusion/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 182, in forward
    feature = self.extract_lidar_features(points)
  File "/home/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 125, in extract_lidar_features
    x = self.encoders["lidar"]["backbone"](feats, coords, batch_size, sizes=sizes)
  File "/home/anaconda3/envs/fusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/anaconda3/envs/fusion/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/bevfusion/mmdet3d/models/backbones/sparse_encoder.py", line 116, in forward
    x = self.conv_input(input_sp_tensor)
  File "/home/anaconda3/envs/fusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/bevfusion/mmdet3d/ops/spconv/modules.py", line 132, in forward
    input = module(input)
  File "/home/anaconda3/envs/fusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/bevfusion/mmdet3d/ops/spconv/conv.py", line 163, in forward
    outids, indice_pairs, indice_pair_num = ops.get_indice_pairs(
  File "/home/bevfusion/mmdet3d/ops/spconv/ops.py", line 92, in get_indice_pairs
    return get_indice_pairs_func(
RuntimeError: mmdet3d/ops/spconv/src/indice_cuda.cu 124
cuda execution failed with error 2

could you help me fix this?

kentang-mit, 2022-08-22:
This is very possibly an OOM problem. I posted some solution [here](https://github.com/mit-han-lab/bevfusion/issues/26) to slightly reduce the memory consumption. We will release a new version of this codebase that requires much less memory for both training and inference in the future, please stay tuned.

TheFatBlue, 2022-08-24:
I got it, thanks a lot. and looking forward to your new work!

closed 2022-08-24 by TheFatBlue

Son-Goku-gpu, 2022-09-01:
Hi @TheFatBlue Did you resolve the issue "cuda execution failed with error 2" above? Thanks!

TheFatBlue, 2022-09-02:
> Hi @TheFatBlue Did you resolve the issue "cuda execution failed with error 2" above? Thanks!

No, I haven't. The solution [here](https://github.com/mit-han-lab/bevfusion/issues/26) didn't work, and I think it's because the memory of the GPU I use(M40) is not large enough. I will have a try on other GPU in the near future