mit-han-lab/bevfusion #110: Waymo Performance and Code

agent-sgs, 2022-08-21:
This is a nice work. Would you release the code and performance evaluation on the Waymo Open Dataset?
I expect your positive response.

kentang-mit, 2022-08-21:
Yes. We will release some experiment code on Waymo. We are working on code cleanning now.

Son-Goku-gpu, 2022-08-23:
@kentang-mit Thanks for your open-source code! May I know if you would release both training&evaluation code and pre-trained models in this week?

Son-Goku-gpu, 2022-08-23:
@kentang-mit I see. I'd like to incoporate your model on Waymo into my framework, would you like to give a rough time for releasing your code? Thanks a lot!

kentang-mit, 2022-08-23:
@Son-Goku-gpu Glad to here that! we are working hard towards a new release in mid September.

kentang-mit, 2022-09-06:
Closed due to inactivity.

closed 2022-09-06 by kentang-mit

Rtakaha, 2022-09-21:
Hi, I am also looking forward to your code for Waymo Open Dataset. It will be great :)

I don't want to rush you, but would you tell me if the code is going to be released soon? Or does it take a little more time?

Thank you very much for your great work!