mit-han-lab/bevfusion #111: No such file or directory: ':/usr/local/cuda-11.0:/usr/local/cuda-11.0/bin/nvcc'

oneway3124, 2022-08-22:
(bevfusion) root@ea6778bb3974:~/bevfusion# python setup.py develop
running develop
/root/miniconda3/envs/bevfusion/lib/python3.8/site-packages/setuptools/command/easy_install.py:144: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/root/miniconda3/envs/bevfusion/lib/python3.8/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
running egg_info
writing mmdet3d.egg-info/PKG-INFO
writing dependency_links to mmdet3d.egg-info/dependency_links.txt
writing top-level names to mmdet3d.egg-info/top_level.txt
/root/miniconda3/envs/bevfusion/lib/python3.8/site-packages/torch/utils/cpp_extension.py:381: UserWarning: Attempted to use ninja as the BuildExtension backend but we could not find ninja.. Falling back to using the slow distutils backend.
  warnings.warn(msg.format('we could not find ninja.'))
reading manifest file 'mmdet3d.egg-info/SOURCES.txt'
adding license file 'LICENSE'
writing manifest file 'mmdet3d.egg-info/SOURCES.txt'
running build_ext
error: [Errno 2] No such file or directory: ':/usr/local/cuda-11.0:/usr/local/cuda-11.0/bin/nvcc'
(bevfusion) root@ea6778bb3974:~/bevfusion#

oneway3124, 2022-08-22:
Hi Sir,

Can you tell me why?

oneway3124, 2022-08-23:
(base) root@8cd1c5a2b9cd:~# /usr/local/cuda
cuda/      cuda-11.0/

oneway3124, 2022-08-23:
Yes, I check it.

oneway3124, 2022-08-23:
[image]

[image]

kentang-mit, 2022-08-23:
I see. What about pointing the `CUDA_HOME` variable to `/usr/local/cuda/` (or `/usr/local/cuda-11.0`)?

oneway3124, 2022-08-23:
Thanks. I have tried these debug methods. But it is also not ok.
So I have switched to another GPU server for this environment.
[image]

oneway3124, 2022-08-23:
That is cuda 10.

kentang-mit, 2022-08-23:
Glad that the problem is resolved on another machine! I'm not sure why CUDA 11.x is not working on your machine since I am also using CUDA 11.x.

oneway3124, 2022-08-24:
It maybe is a reason that the AI cloud machine from "http://gpu.ai-galaxy.cn/".

closed 2022-08-24 by oneway3124

tiancaipipi110, 2023-09-02:
Which server did you switch to ?

tiancaipipi110, 2023-09-02:
@oneway3124  Which server did you switch to?