mit-han-lab/bevfusion #112: training problem

leavesullic, 2022-08-24:
I'm a new rookie to be honest, and I wonder where to modify the batch_size in this project.
I got no result after searched in the config files.
I ask this question because all my 4 V100 gpus use very little memory even I modify the 'samples_per_gpu' to 400, which is very inefficient.
[image]
Thanks for your help and looking forward to your guidance :)

kentang-mit, 2022-08-24:
`samples_per_gpu` is the correct field to modify. However, your observation does seem abnormal to me. Perhaps you can add the `-v` flag to the `torchpack` command to see whether there are error messages.

leavesullic, 2022-08-25:
> `samples_per_gpu` is the correct field to modify. However, your observation does seem abnormal to me. Perhaps you can add the `-v` flag to the `torchpack` command to see whether there are error messages.

Thanks a lot, I'll try

leavesullic, 2022-08-25:
Hi Haotian,
I solved last problem by adding the -v flag, thank you very much.
However new problem came out:

Traceback (most recent call last):
  File "tools/train.py", line 96, in <module>
    main()
  File "tools/train.py", line 85, in main
    train_model(
  File "/home/yetonghui/project/mitbevfusion/mmdet3d/apis/train.py", line 93, in train_model
    cfg.lr_config,
  File "/home/yetonghui/project/tools/anaconda3/envs/mit_bevfusion/lib/python3.8/site-packages/mmcv/utils/config.py", line 507, in __getattr__
    return getattr(self._cfg_dict, name)
  File "/home/yetonghui/project/tools/anaconda3/envs/mit_bevfusion/lib/python3.8/site-packages/mmcv/utils/config.py", line 48, in __getattr__
    raise ex
AttributeError: 'ConfigDict' object has no attribute 'lr_config'

I checked the items in cfg but there's no key named 'lr_config'. So I wonder where to find this 'lr-config'. 
Maybe I got some omissions in the yaml files? But I git clone this project completely, which means the yaml files are completely correct.
Please point out my mistakes.

Best,
Tonghui
[image]
[image]

kentang-mit, 2022-08-25:
That's because we haven't released the `lr_config` in the current release. We anticipate that the training details will be released next month, please stay tuned.

leavesullic, 2022-08-26:
> 

thank u :)
It is really an excellent work!

wish u the best

kentang-mit, 2022-09-06:
Closed due to inactivity.