mit-han-lab/bevfusion #113: Issue while training

VeeranjaneyuluToka, 2022-08-26:
Am getting the below error while training with the below command
**torchpack dist-run -np 1 -v python tools/train.py configs/nuscenes/det/centerhead/lssfpn/camera/256x704/swint/default.yaml**

  File "tools/train.py", line 78, in main
    train_model(
  File "/home/hykeserver/Downloads/bevfusion/mmdet3d/apis/train.py", line 127, in train_model
    runner.run(data_loaders, [("train", 1)])
  File "/home/hykeserver/anaconda3/envs/bevf_ptt19/lib/python3.8/site-packages/mmcv/runner/epoch_based_runner.py", line 127, in run
    epoch_runner(data_loaders[i], **kwargs)
  File "/home/hykeserver/Downloads/bevfusion/mmdet3d/runner/epoch_based_runner.py", line 14, in train
    super().train(data_loader, **kwargs)
  File "/home/hykeserver/anaconda3/envs/bevf_ptt19/lib/python3.8/site-packages/mmcv/runner/epoch_based_runner.py", line 51, in train
    self.call_hook('after_train_iter')
  File "/home/hykeserver/anaconda3/envs/bevf_ptt19/lib/python3.8/site-packages/mmcv/runner/base_runner.py", line 307, in call_hook
    getattr(hook, fn_name)(self)
  File "/home/hykeserver/anaconda3/envs/bevf_ptt19/lib/python3.8/site-packages/mmcv/runner/hooks/optimizer.py", line 228, in after_train_iter
    grad_norm = self.clip_grads(runner.model.parameters())
  File "/home/hykeserver/anaconda3/envs/bevf_ptt19/lib/python3.8/site-packages/mmcv/runner/hooks/optimizer.py", line 31, in clip_grads
    return clip_grad.clip_grad_norm_(params, **self.grad_clip)
TypeError: clip_grad_norm_() argument after ** must be a mapping, not str

Any help would be really helpful.

kentang-mit, 2022-08-26:
You may try adding this to the config:

```yaml
optimizer_config:
  grad_clip:
    max_norm: 35
    norm_type: 2
```

VeeranjaneyuluToka, 2022-08-29:
Thanks, that works.

closed 2022-08-29 by VeeranjaneyuluToka