mit-han-lab/bevfusion #114: Can not output files for visualization.

suker799, 2022-08-29:
Thanks for your amazing work!
When I use the command as the format " torchpack dist-run -np 1 python tools/test.py ${CONFIG_FILE} ${CKPT_PATH} --out ${RESULTS_PATH} --eval 'mAP' --eval-options 'show=False' 'out_dir=${SHOW_DIR}' ", it can correctly output the value mAP and the result file *.pkl except the files for visulization in the directory "${SHOW_DIR}". How can I get the input data, the output of networks and ground-truth labels visualized on the input (e.g. ***_points.obj, ***_pred.obj, ***_gt.obj, ***_img.png and ***_pred.png in multi-modality detection task) in ${SHOW_DIR}. Thanks a lot. 

The detailed command is shown as below, 
torchpack dist-run -np 1 python tools/test.py /home/lvning/project/bev-test/configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml /home/lvning/project/bev-test/pretrained/bevfusion-det.pth --eval 'mAP' --out '/data_disk/lvning/output/image1/result/image1.pkl' --eval-options 'show=False' 'out_dir=/data_disk/lvning/output/image1/show'

kentang-mit, 2022-08-30:
Hello @suker799,

Thanks for your interest. This is not the correct way to use our visualization script. Please refer to [this issue](https://github.com/mit-han-lab/bevfusion/issues/59) for the visualization command. You might need to adjust the `box-score` threshold.

Best,
Haotian

kentang-mit, 2022-09-06:
Closed due to inactivity.

closed 2022-09-06 by kentang-mit