mit-han-lab/bevfusion #115: Evaluation problem

lilkeker, 2022-08-29:
Traceback (most recent call last):
  File "tools/test.py", line 230, in <module>
    main()
  File "tools/test.py", line 180, in main
    model = build_model(cfg.model, test_cfg=cfg.get("test_cfg"))
  File "/home/lk/bevfusion/mmdet3d/models/builder.py", line 41, in build_model
    return build_fusion_model(cfg, train_cfg=train_cfg, test_cfg=test_cfg)
  File "/home/lk/bevfusion/mmdet3d/models/builder.py", line 36, in build_fusion_model
    cfg, default_args=dict(train_cfg=train_cfg, test_cfg=test_cfg)
  File "/home/lk/anaconda3/envs/uvtr/lib/python3.7/site-packages/mmcv/utils/registry.py", line 212, in build
    return self.build_func(*args, **kwargs, registry=self)
  File "/home/lk/anaconda3/envs/uvtr/lib/python3.7/site-packages/mmcv/utils/registry.py", line 55, in build_from_cfg
    raise type(e)(f'{obj_cls.__name__}: {e}')
TypeError: __init__() missing 4 required positional arguments: 'code', 'msg', 'hdrs', and 'fp'

How to solve it?

kentang-mit, 2022-08-30:
Hi @lilkeker,

I would suggest you to have a look at the environment. Are you using `mmcv==1.4.0` and `mmdetection==2.20.0`? 

Best,
Haotian

kentang-mit, 2022-09-06:
Closed due to inactivity. Feel free to reopen if the problem still exists.

closed 2022-09-06 by kentang-mit