mit-han-lab/bevfusion #116: Dataset nuscenes issue

oneway3124, 2022-08-29:
When I use the following to process my full dataset of nuscenes:
python tools/create_data.py nuscenes --root-path ./data/nuscenes --out-dir ./data/nuscenes --extra-tag nuscenes

There is no following files,
│   │   ├── nuscenes_infos_train_mono3d.coco.json
│   │   ├── nuscenes_infos_val_mono3d.coco.json
│   │   ├── nuscenes_infos_test_mono3d.coco.json
Only having,
├── data
│   ├── nuscenes
│   │   ├── maps
│   │   ├── samples
│   │   ├── sweeps
│   │   ├── v1.0-test
|   |   ├── v1.0-trainval
│   │   ├── nuscenes_database
│   │   ├── nuscenes_infos_train.pkl
│   │   ├── nuscenes_infos_val.pkl
│   │   ├── nuscenes_infos_test.pkl
│   │   ├── nuscenes_dbinfos_train.pkl

But have one new folder, name as nuscenes_gt_database.

The link which I refer to is https://github.com/open-mmlab/mmdetection3d/blob/master/docs/en/datasets/nuscenes_det.md .

kentang-mit, 2022-08-30:
Hi @oneway3124,

I think this is OK. We currently do not need those info files for monocular detection. I'll remove them in the README to avoid confusion.

Best,
Haotian

kentang-mit, 2022-09-24:
Closed due to inactivity.

closed 2022-09-24 by kentang-mit