mit-han-lab/bevfusion #117: Stuck in build_model

dvando, 2022-08-30:
Hi, I love your work. Thanks for sharing the code.

I've encountered a problem when trying to test with nuScenes data, the program is stuck in build_model. 
It stuck right after the load checkpoint:

`2022-08-29 13:45:06,429 - mmdet - INFO - load checkpoint from http path: https://github.com/SwinTransformer/storage/releases/download/v1.0.0/swin_tiny_patch4_window7_224.pth`

I'm using the same exact version of every library with the prerequisites.
I tried to use the test with other modalities, but all of them gave the same result, stuck right after its loading checkpoint.
Any idea why is it happening? I found out that the stuck is happening in build_model by printing something after each process.

Thanks in advance!

dvando, 2022-08-30:
I tried using the `-v` flag and it shows that I have an error in the CUDA set device. I'll try something about it in the meantime, but I would still like any tips and advice from anyone, thanks.

[image]

dvando, 2022-08-30:
Okay, I just found out that -np flag in torchpack is the total GPU that I have. My bad, guys!

closed 2022-08-30 by dvando

DKandrew, 2023-08-14:
Thanks for figuring out this issue. This saves my day :)