mit-han-lab/bevfusion #118: ImportError: cannot import name 'recursive_eval' from 'mmdet3d.utils' (/home/disk01/wyw/mmdetection3d/mmdet3d/utils/__init__.py)

wang911205, 2022-08-30:
ubuntu16:~/mmdetection3d/bevfusion$ python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth --eval bbox
Traceback (most recent call last):
  File "tools/test.py", line 20, in <module>
    from mmdet3d.utils import recursive_eval
ImportError: cannot import name 'recursive_eval' from 'mmdet3d.utils' (/home/disk01/wyw/mmdetection3d/mmdet3d/utils/__init__.py)

kentang-mit, 2022-08-30:
Hi @wang911205,

Please make sure that you have run 

```bash
python setup.py develop
```

under our root folder.

Best,
Haotian

Aruen24, 2022-09-01:
> Hi @wang911205,
> 
> Please make sure that you have run
> 
> ```shell
> python setup.py develop
> ```
> 
> under our root folder.
> 
> Best, Haotian
 After run python setup.py develop

nvcc fatal   : Unsupported gpu architecture 'compute_80'
error: command '/usr/local/cuda-10.2/bin/nvcc' failed with exit code 1

kentang-mit, 2022-09-01:
May I have the information about your PyTorch version and GPU type? You probably will need to use PyTorch 1.10.2 to circumvent the THC-related error. 

For the nvcc fatal error, this is because your CUDA devkit is not new enough to support compute_80. As such, I would recommend you to change these two lines to the compute capability of your device accordingly. For example, GTX1080Ti is compute_61 and sm_61, Tesla V100 is compute_70 and sm_70, RTX2080Ti is compute_75 and sm_75.

Aruen24, 2022-09-01:
> 
torch1.11.0
Tesla P40
cuda10.2

Aruen24, 2022-09-01:
> May I have the information about your PyTorch version and GPU type? You probably will need to use PyTorch 1.10.2 to circumvent the THC-related error.
>
> For the nvcc fatal error, this is because your [...]

torch1.11.0
Tesla P40
cuda10.2

kentang-mit, 2022-09-01:
As I have mentioned, you probably have to downgrade your PyTorch to 1.10.2 temporarily. Tesla P40 has compute capability compute_61 (and sm_61). Please modify the setup file accordingly.

Aruen24, 2022-09-07:
> As I have mentioned, you probably have to downgrade your PyTorch to 1.10.2 temporarily. Tesla P40 has compute capability compute_61 (and sm_61). Please modify the setup file accordingly.

 I downgrade my PyTorch to 1.10.1. when I run the test.py there is a new error.

torchpack dist-run -np 1 python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth --eval bbox
ssh: Could not resolve hostname localhost:1: Name or service not known
--------------------------------------------------------------------------
ORTE was unable to reliably start one or more daemons.
This usually is caused by:

* not finding the required libraries and/or binaries on
  one or more nodes. Please check your PATH and LD_LIBRARY_PATH
  settings, or configure OMPI with --enable-orterun-prefix-by-default

* lack of authority to execute on one or more specified nodes.
  Please verify your allocation and authorities.

* the inability to write startup files into /tmp (--tmpdir/orte_tmpdir_base).
  Please check with your sys admin to determine the correct location to use.

*  compilation of the orted with dynamic libraries when static are required
  (e.g., on Cray). Please check your configure cmd line and consider using
  one of the contrib/platform definitions for your system type.

* an inability to create a connection back to mpirun due to a
  lack of common network interfaces and/or no route found between
  them. Please check network connectivity (including firewalls
  and network routing requirements).

kentang-mit, 2022-09-08:
Did you configure OpenMPI correctly? There are some directions in [this issue](https://github.com/mit-han-lab/bevfusion/issues/21), you may have a look.

kentang-mit, 2022-09-24:
Closed due to inactivity.

closed 2022-09-24 by kentang-mit