mit-han-lab/bevfusion #119: Spconv Bug

Son-Goku-gpu, 2022-08-31:
Hi, do you have any suggestion for the bug below? Thanks!

[image]

kentang-mit, 2022-08-31:
This might be an OOM problem. You may check out the [solution](https://github.com/mit-han-lab/bevfusion/issues/26) I posted in this issue related to reversing the order of features.

Son-Goku-gpu, 2022-09-01:
I guess it may be related with the cuda, may I know your cuda version for your development?

kentang-mit, 2022-09-01:
Sure. We use CUDA 11.3. I think CUDA 11.x will be fine as long as it matches the version that PyTorch is compiled with.

Son-Goku-gpu, 2022-09-01:
em...All my softwares are installed with cuda version 11.1:
```
Python 3.7
OpenMPI 4.0.4 and mpi4py 3.1.3
Pillow 8.4.0
PyTorch 1.10.1
tqdm 4.42.1
torchpack 0.3.1
mmcv-full 1.4.0
mmdetection 2.25.1
nuscenes-dev-kit latest
```
and still met the same problem. Any advice?

kentang-mit, 2022-09-01:
For the current codebase it is possible that the required memory is larger than 12G. We will publish a new version of the codebase that consumes less GPU memory during training & testing. The testing memory requirement will be much smaller than 12G after that new release. 

Note: one way to confirm whether it is an OOM problem is that you can try the inference for single-modality models.

Son-Goku-gpu, 2022-09-02:
@kentang-mit Thanks. I guess it should be OOM problem, as I tried to replace ops/spconv with sp2.x and it works.

kentang-mit, 2022-09-04:
Hi @Son-Goku-gpu,

Glad to hear that. I'm closing this issue temporarily because the problem is released. Our new release will actually be based on spconv 2.x.

Best,
Haotian

closed 2022-09-04 by kentang-mit