mit-han-lab/bevfusion #120: Is there any command to download all the dependencies?

Richard-LYF, 2022-08-31:
Is there any command to download all the dependencies?

kentang-mit, 2022-08-31:
I currently did not have one off-the-shelf. For most of the dependencies you can simply run `pip install [library name]`, where the library name is given in the **Prerequisites** section of our README. For mmcv, please remember to use the following command:

```bash
pip install mmcv==1.4.0 mmcv-full==1.4.0
```

kentang-mit, 2022-09-24:
Closed due to inactivity. 

Update: docker support is provided in these two PRs: [144](https://github.com/mit-han-lab/bevfusion/pull/144) and [145](https://github.com/mit-han-lab/bevfusion/pull/145). You are welcomed to have a look at them.

closed 2022-09-24 by kentang-mit