mit-han-lab/bevfusion #122: Test error

X-ray0214, 2022-08-31:
When I run "python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth --eval bbox", I got error as follows:

completed: 0, elapsed: 0sTraceback (most recent call last):
  File "tools/test.py", line 249, in <module>
    main()
  File "tools/test.py", line 245, in main
    print(dataset.evaluate(outputs, **eval_kwargs))
  File "/home1/ugv/autodriving/bevfusion/mmdet3d/datasets/nuscenes_dataset.py", line 544, in evaluate
    if "masks_bev" in results[0]:
IndexError: list index out of range

I printed the "results" and I found it was [] . I checked the dataset and model. They both exist.

How do I solve it?

Hope for your  reply.

kentang-mit, 2022-08-31:
Hello @X-ray0214,

Did you use `torchpack` to launch the command? Are you able to finish the model inference on all samples?

Best,
Haotian

kentang-mit, 2022-09-24:
Closed due to inactivity.

closed 2022-09-24 by kentang-mit