mit-han-lab/bevfusion #124: Train problems

chenrui17, 2022-09-01:
1. train with `torchpack dist-run -np 1 python tools/train.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml`, it run about 150 steps, and occurs error like this: 
```
2022-09-01 13:12:52,847 - mmdet3d - INFO - Epoch [1][50/308]	lr: 1.000e-04, eta: 3:01:59, time: 1.787, data_time: 0.135, memory: 17612, loss/object/loss_heatmap: 572.3525, loss/object/layer_-1_loss_cls: 7.0433, loss/object/layer_-1_loss_bbox: 11.3112, stats/object/matched_ious: 0.0030, loss: 590.7069, grad_norm: nan
2022-09-01 13:14:12,089 - mmdet3d - INFO - Epoch [1][100/308]	lr: 1.000e-04, eta: 2:50:17, time: 1.585, data_time: 0.021, memory: 17612, loss/object/loss_heatmap: 4.1159, loss/object/layer_-1_loss_cls: 4.9029, loss/object/layer_-1_loss_bbox: 10.6588, stats/object/matched_ious: 0.0058, loss: 19.6776, grad_norm: 28.4252
2022-09-01 13:15:33,244 - mmdet3d - INFO - Epoch [1][150/308]	lr: 1.000e-04, eta: 2:46:46, time: 1.623, data_time: 0.024, memory: 17612, loss/object/loss_heatmap: 3.0922, loss/object/layer_-1_loss_cls: 3.4564, loss/object/layer_-1_loss_bbox: 10.4424, stats/object/matched_ious: 0.0085, loss: 16.9910, grad_norm: 23.0907
Traceback (most recent call last):
  File "tools/train.py", line 87, in <module>
    main()
  File "tools/train.py", line 76, in main
    train_model(
  File "/home/users/chenrui17/3d_models/bevfusion-main/mmdet3d/apis/train.py", line 126, in train_model
    runner.run(data_loaders, [("train", 1)])
  File "/opt/conda/envs/bevfusion/lib/python3.8/site-packages/mmcv/runner/epoch_based_runner.py", line 127, in run
    epoch_runner(data_loaders[i], **kwargs)
  File "/home/users/chenrui17/3d_models/bevfusion-main/mmdet3d/runner/epoch_based_runner.py", line 14, in train
    super().train(data_loader, **kwargs)
  File "/opt/conda/envs/bevfusion/lib/python3.8/site-packages/mmcv/runner/epoch_based_runner.py", line 50, in train
    self.run_iter(data_batch, train_mode=True, **kwargs)
  File "/opt/conda/envs/bevfusion/lib/python3.8/site-packages/mmcv/runner/epoch_based_runner.py", line 29, in run_iter
    outputs = self.model.train_step(data_batch, self.optimizer,
  File "/opt/conda/envs/bevfusion/lib/python3.8/site-packages/mmcv/parallel/distributed.py", line 52, in train_step
    output = self.module.train_step(*inputs[0], **kwargs[0])
  File "/home/users/chenrui17/3d_models/bevfusion-main/mmdet3d/models/fusion_models/base.py", line 78, in train_step
    losses = self(**data)
  File "/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/opt/conda/envs/bevfusion/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/users/chenrui17/3d_models/bevfusion-main/mmdet3d/models/fusion_models/bevfusion.py", line 185, in forward
    feature = self.extract_lidar_features(points)
  File "/opt/conda/envs/bevfusion/lib/python3.8/site-packages/nvtx/nvtx.py", line 101, in inner
    result = func(*args, **kwargs)
  File "/home/users/chenrui17/3d_models/bevfusion-main/mmdet3d/models/fusion_models/bevfusion.py", line 128, in extract_lidar_features
    x = self.encoders["lidar"]["backbone"](feats, coords, batch_size, sizes=sizes)
  File "/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/opt/conda/envs/bevfusion/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/users/chenrui17/3d_models/bevfusion-main/mmdet3d/models/backbones/sparse_encoder.py", line 120, in forward
    x = encoder_layer(x)
  File "/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/users/chenrui17/3d_models/bevfusion-main/mmdet3d/ops/spconv/modules.py", line 132, in forward
    input = module(input)
  File "/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/users/chenrui17/3d_models/bevfusion-main/mmdet3d/ops/spconv/modules.py", line 132, in forward
    input = module(input)
  File "/opt/conda/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/users/chenrui17/3d_models/bevfusion-main/mmdet3d/ops/spconv/conv.py", line 163, in forward
    outids, indice_pairs, indice_pair_num = ops.get_indice_pairs(
  File "/home/users/chenrui17/3d_models/bevfusion-main/mmdet3d/ops/spconv/ops.py", line 92, in get_indice_pairs
    return get_indice_pairs_func(
RuntimeError: /home/users/chenrui17/3d_models/bevfusion-main/mmdet3d/ops/spconv/include/tensorview/helper_launch.h 17
N > 0 assert faild. CUDA kernel launch blocks must be positive, but got N= 0
--------------------------------------------------------------------------
Primary job  terminated normally, but 1 process returned
a non-zero exit code. Per user-direction, the job has been aborted.
--------------------------------------------------------------------------
--------------------------------------------------------------------------
mpirun detected that one or more processes exited with non-zero status, thus causing
the job to be terminated. The first process to do so was:

  Process name: [[39935,1],0]
  Exit code:    1
```

2. train with `torchpack dist-run -np 8 python tools/train.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml` , when start trainning, it hangs and by `nvidia-smi`  ,it looks like gpu stuck.

my install step is 
```
conda create -n bevfusion python=3.8
source activate bevfusion
conda install pytorch==1.10.1 torchvision==0.11.2 torchaudio==0.10.1 cudatoolkit=11.3 -c pytorch -c conda-forge
pip install mmcv==1.4.0
pip install mmcv-full==1.4.0  -f https://download.openmmlab.com/mmcv/dist/cu113/torch1.10.0/index.html
pip install torchpack
pip install mmdet
pip install numba
pip install nuscenes-devkit
pip install mpi4py
python setup.py develop
```

Richard-LYF, 2022-09-01:
did you download OpenMPI = 4.0.4 ?

kentang-mit, 2022-09-01:
We currently haven't released all details about training. This might be related to the training schedule. Please stay tuned, we will release the training code this month.

Richard-LYF, 2022-09-12:
when the traing code will be released?

HengLoveYi, 2022-09-22:
> We currently haven't released all details about training. This might be related to the training schedule. Please stay tuned, we will release the training code this month.

will be appreciated to wait for your train code.