mit-han-lab/bevfusion #127: nvcc fatal : Unsupported gpu architecture 'compute_80'

usmanraza121, 2022-09-03:
after running "**python setup.py develop"** I faced the above error.

**error: command '/usr/local/cuda/bin/nvcc' failed with exit code 1**

currently, I have installed Cuda-10.2. however, if run the above command with Cuda-11.7 or 9.2 it gives an error and suggests installing Cuda-10.2. 
can it only be compiled with Cuda-10.2? no other Cuda-version?

Thank you

MayLkn, 2022-09-04:
> 

Cuda11.1 can also，

kentang-mit, 2022-09-04:
Hello folks,

Would you mind sharing the GPUs you are using? The reason why you meet with this error is that `compute_80` and `sm_80` is too new for your nvcc compiler. You can simply remove the corresponding line in setup.py.

Best,
Haotian

usmanraza121, 2022-09-04:
@kentang-mit  Thank you for your prompt response. I'm using GTX TITAN X GPU (12GB).

usmanraza121, 2022-09-04:
> > 
> 
> Cuda11.1 can also，

Thank you

kentang-mit, 2022-09-04:
Hi @usmanraza121,

In this case, you can replace that line with `compute_61` and `sm_61`.

Best,
Haotian

kentang-mit, 2022-09-24:
Closed due to inactivity.

closed 2022-09-24 by kentang-mit