mit-han-lab/bevfusion #128: A question about box center origin

lzhbrian, 2022-09-04:
Hi,

I notice that you change box origin from `(0.5, 0.5, 0.5)` to `(0.5, 0.5, 0)` at [here](https://github.com/mit-han-lab/bevfusion/blob/6908c57b7800268a9cdf3efeacbfb886d6077d6f/mmdet3d/datasets/nuscenes_dataset.py#L314). 
Since Nuscenes' boxes are originated from `(0.5, 0.5, 0.5)` (as in [here](https://github.com/open-mmlab/mmdetection3d/blob/9556958fe1c6fe432d55a9f98781b8fdd90f4e9c/mmdet3d/datasets/nuscenes_dataset.py#L293) in the original mmdet3d), why did you change it to `(0.5, 0.5, 0)` ?  I am not very familiar with mmdet3d codebase, please correct me if I have understood anything wrong.

Thanks!

kentang-mit, 2022-09-04:
I remember that is related to some of our early explorations on camera-based perception models. The prediction on z dimension will not influence the metrics on nuScenes, so it should also be OK to use (0.5, 0.5, 0.5).

kentang-mit, 2022-09-24:
Closed due to inactivity.

U201310646, 2023-01-06:
Hi,
the same question but i wanna ask if this change  will have an impact on the lidar's data augmentation? 
> https://github.com/mit-han-lab/bevfusion/blob/6908c57b7800268a9cdf3efeacbfb886d6077d6f/mmdet3d/datasets/nuscenes_dataset.py#L314-L317
origin set to (0.5,0.5,0), the following code wont be executed, the center is still geometric center of b-box
> https://github.com/mit-han-lab/bevfusion/blob/3d91980d21556bf697f51b717875d0a31e5be2dd/mmdet3d/core/bbox/structures/base_box3d.py#L61-L64      

but i noticed that  the origin here (0.5,0.5,0) in points_in_rbbox function, the center is considered as center of bottom surface to caculate the corners. Is this inconsistent?
> https://github.com/mit-han-lab/bevfusion/blob/3d91980d21556bf697f51b717875d0a31e5be2dd/mmdet3d/core/bbox/box_np_ops.py#L430