mit-han-lab/bevfusion #129: How to use your own data

MayLkn, 2022-09-06:
Hello author, how to put their own data into six videos

kentang-mit, 2022-09-06:
Hello @MayLkn,

To inference with your custom data, please make sure that you pass in arguments in [lines 149-163](https://github.com/mit-han-lab/bevfusion/blob/e4d599edd51f758fdbf1f6a58732d31c6f8a56cc/mmdet3d/models/fusion_models/bevfusion.py#L149) to the model. You may refer to our [dataset file](https://github.com/mit-han-lab/bevfusion/blob/main/mmdet3d/datasets/nuscenes_dataset.py) for more details about how to get these arguments. 

Best,
Haotian

MayLkn, 2022-09-07:
Which PY file needs to be modified

VeeranjaneyuluToka, 2022-09-07:
Hi, if i understand correctly, you need to create your own py file for your own dataset the way nuscenes_dataset.py gets created for nuscenes dataset as he pointed out in second blue text.

MayLkn, 2022-09-07:
> ，如果我的理解，你需要为自己的数据集创建自己的蓝色 py 文件，就像他在第二个文本中的那样，为 nuscenes_dataset.py 正确为 nuscenes 数据集创建。

Thank you for your reply. It will be difficult

kentang-mit, 2022-09-24:
Closed due to inactivity.

closed 2022-09-24 by kentang-mit