mit-han-lab/bevfusion #131: RuntimeError: CUDA error: __global__ function call is not configured

RG2806, 2022-09-06:
hey, awesome work.
i tried running the evaluation but got this
Traceback (most recent call last):
  File "tools/test.py", line 236, in <module>
    main()
  File "tools/test.py", line 202, in main
    outputs = single_gpu_test(model, data_loader)
  File "/home/predev/repos/amir/bevfusion-main/mmdet3d/apis/test.py", line 15, in single_gpu_test
    result = model(return_loss=False, rescale=True, **data)
  File "/home/predev/repos/amir/bevfusion-main/venv/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/predev/repos/amir/bevfusion-main/venv/lib/python3.8/site-packages/mmcv/parallel/data_parallel.py", line 50, in forward
    return super().forward(*inputs, **kwargs)
  File "/home/predev/repos/amir/bevfusion-main/venv/lib/python3.8/site-packages/torch/nn/parallel/data_parallel.py", line 166, in forward
    return self.module(*inputs[0], **kwargs[0])
  File "/home/predev/repos/amir/bevfusion-main/venv/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/predev/repos/amir/bevfusion-main/venv/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/predev/repos/amir/bevfusion-main/mmdet3d/models/fusion_models/bevfusion.py", line 182, in forward
    feature = self.extract_lidar_features(points)
  File "/home/predev/repos/amir/bevfusion-main/mmdet3d/models/fusion_models/bevfusion.py", line 123, in extract_lidar_features
    feats, coords, sizes = self.voxelize(x)
  File "/home/predev/repos/amir/bevfusion-main/venv/lib/python3.8/site-packages/torch/autograd/grad_mode.py", line 28, in decorate_context
    return func(*args, **kwargs)
  File "/home/predev/repos/amir/bevfusion-main/venv/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 214, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/predev/repos/amir/bevfusion-main/mmdet3d/models/fusion_models/bevfusion.py", line 133, in voxelize
    f, c, n = self.encoders["lidar"]["voxelize"](res)
  File "/home/predev/repos/amir/bevfusion-main/venv/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/predev/repos/amir/bevfusion-main/mmdet3d/ops/voxel/voxelize.py", line 131, in forward
    return voxelization(
  File "/home/predev/repos/amir/bevfusion-main/mmdet3d/ops/voxel/voxelize.py", line 55, in forward
    voxel_num = hard_voxelize(
RuntimeError: CUDA error: __global__ function call is not configured
--------------------------------------------------------------------------
Primary job  terminated normally, but 1 process returned
a non-zero exit code. Per user-direction, the job has been aborted.
--------------------------------------------------------------------------
--------------------------------------------------------------------------
mpirun detected that one or more processes exited with non-zero status, thus causing
the job to be terminated. The first process to do so was:

  Process name: [[40258,1],1]
  Exit code:    134
--------------------------------------------------------------------------

kentang-mit, 2022-09-06:
Hi @RG2806,

That error looks strange to me. Are you able to run the 

```bash
python setup.py develop
```
command without any error?

Best,
Haotian

RG2806, 2022-09-07:
hi @kentang-mit,
here is the output of **python setup.py develop**
running develop
running egg_info
writing mmdet3d.egg-info/PKG-INFO
writing dependency_links to mmdet3d.egg-info/dependency_links.txt
writing top-level names to mmdet3d.egg-info/top_level.txt
reading manifest file 'mmdet3d.egg-info/SOURCES.txt'
writing manifest file 'mmdet3d.egg-info/SOURCES.txt'
running build_ext
building 'mmdet3d.ops.spconv.sparse_conv_ext' extension
Emitting ninja build file /home/predev/repos/amir/bevfusion-main/build/temp.linux-x86_64-3.8/build.ninja...
Compiling objects...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
g++ -pthread -shared -B /home/predev/anaconda3/compiler_compat -L/home/predev/anaconda3/lib -Wl,-rpath=/home/predev/anaconda3/lib -Wl,--no-as-needed -Wl,--sysroot=/ /home/predev/repos/amir/bevfusion-main/build/temp.linux-x86_64-3.8/mmdet3d/ops/spconv/src/all.o /home/predev/repos/amir/bevfusion-main/build/temp.linux-x86_64-3.8/mmdet3d/ops/spconv/src/reordering.o /home/predev/repos/amir/bevfusion-main/build/temp.linux-x86_64-3.8/mmdet3d/ops/spconv/src/reordering_cuda.o /home/predev/repos/amir/bevfusion-main/build/temp.linux-x86_64-3.8/mmdet3d/ops/spconv/src/indice.o /home/predev/repos/amir/bevfusion-main/build/temp.linux-x86_64-3.8/mmdet3d/ops/spconv/src/indice_cuda.o /home/predev/repos/amir/bevfusion-main/build/temp.linux-x86_64-3.8/mmdet3d/ops/spconv/src/maxpool.o /home/predev/repos/amir/bevfusion-main/build/temp.linux-x86_64-3.8/mmdet3d/ops/spconv/src/maxpool_cuda.o -L/home/predev/repos/amir/bevfusion-main/venv/lib/python3.8/site-packages/torch/lib -L/usr/local/cuda/lib64 -lc10 -ltorch -ltorch_cpu -ltorch_python -lcudart -lc10_cuda -ltorch_cuda_cu -ltorch_cuda_cpp -o build/lib.linux-x86_64-3.8/mmdet3d/ops/spconv/sparse_conv_ext.cpython-38-x86_64-linux-gnu.so
building 'mmdet3d.ops.bev_pool.bev_pool_ext' extension
Emitting ninja build file /home/predev/repos/amir/bevfusion-main/build/temp.linux-x86_64-3.8/build.ninja...
Compiling objects...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
g++ -pthread -shared -B /home/predev/anaconda3/compiler_compat -L/home/predev/anaconda3/lib -Wl,-rpath=/home/predev/anaconda3/lib -Wl,--no-as-needed -Wl,--sysroot=/ /home/predev/repos/amir/bevfusion-main/build/temp.linux-x86_64-3.8/mmdet3d/ops/bev_pool/src/bev_pool.o /home/predev/repos/amir/bevfusion-main/build/temp.linux-x86_64-3.8/mmdet3d/ops/bev_pool/src/bev_pool_cuda.o -L/home/predev/repos/amir/bevfusion-main/venv/lib/python3.8/site-packages/torch/lib -L/usr/local/cuda/lib64 -lc10 -ltorch -ltorch_cpu -ltorch_python -lcudart -lc10_cuda -ltorch_cuda_cu -ltorch_cuda_cpp -o build/lib.linux-x86_64-3.8/mmdet3d/ops/bev_pool/bev_pool_ext.cpython-38-x86_64-linux-gnu.so
building 'mmdet3d.ops.iou3d.iou3d_cuda' extension
Emitting ninja build file /home/predev/repos/amir/bevfusion-main/build/temp.linux-x86_64-3.8/build.ninja...
Compiling objects...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
g++ -pthread -shared -B /home/predev/anaconda3/compiler_compat -L/home/predev/anaconda3/lib -Wl,-rpath=/home/predev/anaconda3/lib -Wl,--no-as-needed -Wl,--sysroot=/ /home/predev/repos/amir/bevfusion-main/build/temp.linux-x86_64-3.8/mmdet3d/ops/iou3d/src/iou3d.o /home/predev/repos/amir/bevfusion-main/build/temp.linux-x86_64-3.8/mmdet3d/ops/iou3d/src/iou3d_kernel.o -L/home/predev/repos/amir/bevfusion-main/venv/lib/python3.8/site-packages/torch/lib -L/usr/local/cuda/lib64 -lc10 -ltorch -ltorch_cpu -ltorch_python -lcudart -lc10_cuda -ltorch_cuda_cu -ltorch_cuda_cpp -o build/lib.linux-x86_64-3.8/mmdet3d/ops/iou3d/iou3d_cuda.cpython-38-x86_64-linux-gnu.so
building 'mmdet3d.ops.voxel.voxel_layer' extension
Emitting ninja build file /home/predev/repos/amir/bevfusion-main/build/temp.linux-x86_64-3.8/build.ninja...
Compiling objects...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
g++ -pthread -shared -B /home/predev/anaconda3/compiler_compat -L/home/predev/anaconda3/lib -Wl,-rpath=/home/predev/anaconda3/lib -Wl,--no-as-needed -Wl,--sysroot=/ /home/predev/repos/amir/bevfusion-main/build/temp.linux-x86_64-3.8/mmdet3d/ops/voxel/src/voxelization.o /home/predev/repos/amir/bevfusion-main/build/temp.linux-x86_64-3.8/mmdet3d/ops/voxel/src/scatter_points_cpu.o /home/predev/repos/amir/bevfusion-main/build/temp.linux-x86_64-3.8/mmdet3d/ops/voxel/src/scatter_points_cuda.o /home/predev/repos/amir/bevfusion-main/build/temp.linux-x86_64-3.8/mmdet3d/ops/voxel/src/voxelization_cpu.o /home/predev/repos/amir/bevfusion-main/build/temp.linux-x86_64-3.8/mmdet3d/ops/voxel/src/voxelization_cuda.o -L/home/predev/repos/amir/bevfusion-main/venv/lib/python3.8/site-packages/torch/lib -L/usr/local/cuda/lib64 -lc10 -ltorch -ltorch_cpu -ltorch_python -lcudart -lc10_cuda -ltorch_cuda_cu -ltorch_cuda_cpp -o build/lib.linux-x86_64-3.8/mmdet3d/ops/voxel/voxel_layer.cpython-38-x86_64-linux-gnu.so
building 'mmdet3d.ops.roiaware_pool3d.roiaware_pool3d_ext' extension
Emitting ninja build file /home/predev/repos/amir/bevfusion-main/build/temp.linux-x86_64-3.8/build.ninja...
Compiling objects...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
[... cut ...]
copying build/lib.linux-x86_64-3.8/mmdet3d/ops/furthest_point_sample/furthest_point_sample_ext.cpython-38-x86_64-linux-gnu.so -> mmdet3d/ops/furthest_point_sample
copying build/lib.linux-x86_64-3.8/mmdet3d/ops/gather_points/gather_points_ext.cpython-38-x86_64-linux-gnu.so -> mmdet3d/ops/gather_points
Creating /home/predev/repos/amir/bevfusion-main/venv/lib/python3.8/site-packages/mmdet3d.egg-link (link to .)
mmdet3d 0.0.0 is already the active version in easy-install.pth

Installed /home/predev/repos/amir/bevfusion-main
Processing dependencies for mmdet3d==0.0.0
Finished processing dependencies for mmdet3d==0.0.0

VeeranjaneyuluToka, 2022-09-07:
Hi, would you mind sharing your setup details here (all the required packages versions) and GPU details. I suspect something wrong with your setup.

kentang-mit, 2022-09-07:
Thank you @VeeranjaneyuluToka for participating in the discussion.

@RG2806 thanks for the information. I am just curious whether it is possible for you to run any model from the official mmdetection3d codebase. For example, the LiDAR-only CenterPoint model?

RG2806, 2022-09-09:
Hi @VeeranjaneyuluToka,
here are the details
python 3.8.3
mpirun (Open MPI) 4.0.3
Pillow 8.4.0
torch 1.10.2+cu113
tqdm 4.64.0
torchpack 0.3.1
mmcv-full 1.4.0
mmdetection3d 2.25.0
NVIDIA-SMI 465.19.01    
Driver Version: 465.19.01    
CUDA Version: 11.3
2 nvidia titan rtx gpu card

RG2806, 2022-09-09:
Hi @kentang-mit, i resolved the error after reinstalling the mmcv library.
Thanks

RG2806, 2022-09-12:
Hey @kentang-mit @VeeranjaneyuluToka 
Found the same error again, apparently it wasn't solved. The model got stuck at the dataloader step. I managed to resolve that, now back with the same error. I tried to get verbose output of the error

Traceback (most recent call last):
  File "tools/test.py", line 236, in <module>
    main()
  File "tools/test.py", line 202, in main
    outputs = single_gpu_test(model, data_loader)
  File "/home/predev/repos/amir/bevfusion-main/mmdet3d/apis/test.py", line 15, in single_gpu_test
    result = model(return_loss=False, rescale=True, **data)
  File "/home/predev/repos/amir/bevfusion-main/venv/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/predev/repos/amir/bevfusion-main/venv/lib/python3.8/site-packages/mmcv/parallel/data_parallel.py", line 50, in forward
    return super().forward(*inputs, **kwargs)
  File "/home/predev/repos/amir/bevfusion-main/venv/lib/python3.8/site-packages/torch/nn/parallel/data_parallel.py", line 166, in forward
    return self.module(*inputs[0], **kwargs[0])
  File "/home/predev/repos/amir/bevfusion-main/venv/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/predev/repos/amir/bevfusion-main/venv/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/predev/repos/amir/bevfusion-main/mmdet3d/models/fusion_models/bevfusion.py", line 169, in forward
    feature = self.extract_camera_features(
  File "/home/predev/repos/amir/bevfusion-main/mmdet3d/models/fusion_models/bevfusion.py", line 108, in extract_camera_features
    x = self.encoders["camera"]["vtransform"](
  File "/home/predev/repos/amir/bevfusion-main/venv/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/predev/repos/amir/bevfusion-main/mmdet3d/models/vtransforms/depth_lss.py", line 100, in forward
    x = super().forward(*args, **kwargs)
  File "/home/predev/repos/amir/bevfusion-main/venv/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 214, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/predev/repos/amir/bevfusion-main/mmdet3d/models/vtransforms/base.py", line 289, in forward
    x = self.bev_pool(geom, x)
  File "/home/predev/repos/amir/bevfusion-main/venv/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 214, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/predev/repos/amir/bevfusion-main/mmdet3d/models/vtransforms/base.py", line 167, in bev_pool
    x = bev_pool(x, geom_feats, B, self.nx[2], self.nx[0], self.nx[1])
  File "/home/predev/repos/amir/bevfusion-main/mmdet3d/ops/bev_pool/bev_pool.py", line 96, in bev_pool
    x = x.permute(0, 4, 1, 2, 3).contiguous()
RuntimeError: CUDA error: an illegal memory access was encountered
terminate called after throwing an instance of 'c10::CUDAError'
  what():  CUDA error: an illegal memory access was encountered
Exception raised from create_event_internal at ../c10/cuda/CUDACachingAllocator.cpp:1211 (most recent call first):
frame #0: c10::Error::Error(c10::SourceLocation, std::string) + 0x42 (0x7f8335a8cd62 in /home/predev/repos/amir/bevfusion-main/venv/lib/python3.8/site-packages/torch/lib/libc10.so)
frame #1: <unknown function> + 0x1c5f3 (0x7f8335cef5f3 in /home/predev/repos/amir/bevfusion-main/venv/lib/python3.8/site-packages/torch/lib/libc10_cuda.so)
frame #2: c10::cuda::CUDACachingAllocator::raw_delete(void*) + 0x1a2 (0x7f8335cf0002 in /home/predev/repos/amir/bevfusion-main/venv/lib/python3.8/site-packages/torch/lib/libc10_cuda.so)
frame #3: c10::TensorImpl::release_resources() + 0xa4 (0x7f8335a76314 in /home/predev/repos/amir/bevfusion-main/venv/lib/python3.8/site-packages/torch/lib/libc10.so)
frame #4: <unknown function> + 0x29eb89 (0x7f83367edb89 in /home/predev/repos/amir/bevfusion-main/venv/lib/python3.8/site-packages/torch/lib/libtorch_python.so)
frame #5: <unknown function> + 0xadfbe1 (0x7f833702ebe1 in /home/predev/repos/amir/bevfusion-main/venv/lib/python3.8/site-packages/torch/lib/libtorch_python.so)
frame #6: THPVariable_subclass_dealloc(_object*) + 0x292 (0x7f833702eee2 in /home/predev/repos/amir/bevfusion-main/venv/lib/python3.8/site-packages/torch/lib/libtorch_python.so)
<omitting python frames>
frame #22: __libc_start_main + 0xf3 (0x7f833cd7f0b3 in /lib/x86_64-linux-gnu/libc.so.6)

[ML007:1492126] *** Process received signal ***
[ML007:1492126] Signal: Aborted (6)
[ML007:1492126] Signal code:  (-6)
[ML007:1492126] [ 0] /lib/x86_64-linux-gnu/libpthread.so.0(+0x153c0)[0x7f833cf5f3c0]
[ML007:1492126] [ 1] /lib/x86_64-linux-gnu/libc.so.6(gsignal+0xcb)[0x7f833cd9e18b]
[ML007:1492126] [ 2] /lib/x86_64-linux-gnu/libc.so.6(abort+0x12b)[0x7f833cd7d859]
[ML007:1492126] [ 3] /lib/x86_64-linux-gnu/libstdc++.so.6(+0x9e911)[0x7f8337e53911]
[ML007:1492126] [ 4] /lib/x86_64-linux-gnu/libstdc++.so.6(+0xaa38c)[0x7f8337e5f38c]
[ML007:1492126] [ 5] /lib/x86_64-linux-gnu/libstdc++.so.6(+0xa9369)[0x7f8337e5e369]
[ML007:1492126] [ 6] /lib/x86_64-linux-gnu/libstdc++.so.6(__gxx_personality_v0+0x2a1)[0x7f8337e5ed21]
[ML007:1492126] [ 7] /home/predev/anaconda3/bin/../lib/libgcc_s.so.1(+0xcaab)[0x7f8339840aab]
[ML007:1492126] [ 8] /home/predev/anaconda3/bin/../lib/libgcc_s.so.1(_Unwind_Resume+0x62)[0x7f8339840f49]
[... cut ...]
--------------------------------------------------------------------------
Primary job  terminated normally, but 1 process returned
a non-zero exit code. Per user-direction, the job has been aborted.
--------------------------------------------------------------------------
--------------------------------------------------------------------------
mpirun noticed that process rank 1 with PID 0 on node ML007 exited on signal 6 (Aborted).
--------------------------------------------------------------------------

kentang-mit, 2022-09-12:
Hi @RG2806,

This might be a different problem. It seems that you are able to launch the customized CUDA kernels, but there are some memory access issues in the kernel itself. Are you able to run inference for the LiDAR-only model?

Best,
Haotian

kentang-mit, 2022-09-14:
That looks strange to me @RG2806. Your error log actually shows that some customized CUDA kernels (e.g. BEV pooling) can run on your machine while the others (such as hard_voxelize) cannot. I looked at your environment again, and the major difference with my setup is the version of mmdetection. Probably you can try installing v2.20.0. Also, it will be better if you haven't installed official mmdetection3d in the current environment.

kentang-mit, 2022-09-24:
Update: docker support is provided in these two PRs: [144](https://github.com/mit-han-lab/bevfusion/pull/144) and [145](https://github.com/mit-han-lab/bevfusion/pull/145). You are welcomed to have a look at them.

kentang-mit, 2022-09-26:
Temporarily closed due to inactivity.

closed 2022-09-26 by kentang-mit