mit-han-lab/bevfusion #133: How much GPU memory when do test?

HengLoveYi, 2022-09-09:
Hello, I would like to know how much cuda memory we need if we use one nvidia GPU to run test.py? 
Best Regards.

VeeranjaneyuluToka, 2022-09-09:
Hi, I was able to run at least camera only model on 8GB gpu and i believe we should be able to run but not tried myself. However i have tried all three models (camera only, lidar only and camera+lidar) in 3090 GPU card which is having 24 gb gpu memory.

zhijian-liu, 2022-09-09:
Thanks, @VeeranjaneyuluToka! An RTX 3090 GPU is sufficient to run the model inference.

closed 2022-09-09 by zhijian-liu

HengLoveYi, 2022-09-14:
Thank you for your answers very much! I have tried and found camer&lidar model need about 16GB to run test.py in a single GPU card.

Deephome, 2022-09-15:
> 

Hi, do you know how to run test.py on a single GPU card? just change 8 to 1 in the following command? 
`torchpack dist-run -np 8 python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth --eval bbox`

VeeranjaneyuluToka, 2022-09-15:
Yes, changing from 8 to 1 should work straight away.

Deephome, 2022-09-16:
> Yes, changing from 8 to 1 should work straight away.

Thank you! I try the following command, 
`torchpack dist-run -np 1 python tools/visualize.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml --mode pred --checkpoint pretrained/bevfusion-det.pth`

but got this erros: 

`Traceback (most recent call last):
  File "tools/visualize.py", line 167, in <module>
    main()
  File "tools/visualize.py", line 89, in main
    outputs = model(**data)
  File "/home/ubuntu/anaconda3/envs/mmlab/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/ubuntu/anaconda3/envs/mmlab/lib/python3.8/site-packages/torch/nn/parallel/distributed.py", line 886, in forward
    output = self.module(*inputs[0], **kwargs[0])
  File "/home/ubuntu/anaconda3/envs/mmlab/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/ubuntu/anaconda3/envs/mmlab/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 98, in new_func
    return old_func(*args, **kwargs)
  File "/home/ubuntu/code/early_fusion/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 182, in forward
    feature = self.extract_lidar_features(points)
  File "/home/ubuntu/code/early_fusion/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 125, in extract_lidar_features
    x = self.encoders["lidar"]["backbone"](feats, coords, batch_size, sizes=sizes)
  File "/home/ubuntu/anaconda3/envs/mmlab/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/ubuntu/anaconda3/envs/mmlab/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 98, in new_func
    return old_func(*args, **kwargs)
  File "/home/ubuntu/code/early_fusion/bevfusion/mmdet3d/models/backbones/sparse_encoder.py", line 120, in forward
    x = encoder_layer(x)
  File "/home/ubuntu/anaconda3/envs/mmlab/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/ubuntu/code/early_fusion/bevfusion/mmdet3d/ops/spconv/modules.py", line 132, in forward
    input = module(input)
  File "/home/ubuntu/anaconda3/envs/mmlab/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/ubuntu/code/early_fusion/bevfusion/mmdet3d/ops/spconv/modules.py", line 132, in forward
    input = module(input)
  File "/home/ubuntu/anaconda3/envs/mmlab/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/ubuntu/code/early_fusion/bevfusion/mmdet3d/ops/spconv/conv.py", line 163, in forward
    outids, indice_pairs, indice_pair_num = ops.get_indice_pairs(
  File "/home/ubuntu/code/early_fusion/bevfusion/mmdet3d/ops/spconv/ops.py", line 92, in get_indice_pairs
    return get_indice_pairs_func(
RuntimeError: /home/ubuntu/code/early_fusion/bevfusion/mmdet3d/ops/spconv/include/tensorview/helper_launch.h 17
N > 0 assert faild. CUDA kernel launch blocks must be positive, but got N= 0`

LeongVan, 2022-12-09:
> 

Hello, have you solved this problem yet?