mit-han-lab/bevfusion #135: Understanding implementation with the paper

VeeranjaneyuluToka, 2022-09-13:
Hi,

I have been looking into Camera-to-BEV transformation, I am looking into the paper and implementation simulteniously (including LSS and BEVDet). Not really spot the point where the implementation is for this in this repo. 

Wanted to use this thread to understand further implementation as well, so would request you to open for some time till i complete understand the code and the paper :). Thanks.

kentang-mit, 2022-09-14:
Hi @VeeranjaneyuluToka,

You can trace the functions in [this file](https://github.com/mit-han-lab/bevfusion/blob/main/mmdet3d/models/vtransforms/lss.py) to understand the implementation.

Best,
Haotian

kentang-mit, 2022-09-24:
Closed due to inactivity.

closed 2022-09-24 by kentang-mit