mit-han-lab/bevfusion #136: When will the code for training the camera only baseline and lidar only baseline be released?

Richard-LYF, 2022-09-15:
When will the code for training the camera only baseline and lidar only baseline be released?

kentang-mit, 2022-09-17:
Thanks for the interest. We are still cleaning the code. Due to some unexpected incident, the code release is slightly delayed. Please stay tuned, thanks for the patience!

IAMShashankk, 2022-09-19:
@kentang-mit Is it possible for you to give us some timeline on this? Earlier it was expected in september. Is it still possbile?

kentang-mit, 2022-09-19:
We are still working hard on it. If eventually we cannot make it, I will release the hyperparameters of these two models you mentioned.

RG2806, 2022-09-19:
@kentang-mit , if releasing only hyperparameters, please do release for Camera+Lidar model as well

kentang-mit, 2022-09-26:
Training details are currently released in [this branch](https://github.com/mit-han-lab/bevfusion/tree/dev/training) and it is under review for merging into the main branch.

For the camera+LiDAR models, I believe it is relatively easy to reproduce the results with the current information we released. We will do a formal code release in the future, together with some memory-efficient features.

closed 2022-09-26 by kentang-mit