mit-han-lab/bevfusion #137: [Errno 2] No such file or directory: ':/usr/local/cuda/bin/nvcc'

BHC1205, 2022-09-16:
Hi,
A same issue which in https://github.com/mit-han-lab/bevfusion/issues/111,
I tried all the metheds, but it does not work,so can you please check to see the reason?

Ubuntu 20.04
Python3.8
pytorch1.10.0+cu113
cuda toolkit 11.3/11.6
nvidia driver 515.65.01
cuda0 : 3090ti

/home/zhwsh/.conda/envs/BEVFusion383/lib/python3.8/site-packages/setuptools/command/easy_install.py:144: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/home/zhwsh/.conda/envs/BEVFusion383/lib/python3.8/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
running egg_info
writing mmdet3d.egg-info/PKG-INFO
writing dependency_links to mmdet3d.egg-info/dependency_links.txt
writing top-level names to mmdet3d.egg-info/top_level.txt
reading manifest file 'mmdet3d.egg-info/SOURCES.txt'
adding license file 'LICENSE'
writing manifest file 'mmdet3d.egg-info/SOURCES.txt'
running build_ext
error: [Errno 2] No such file or directory: ':/usr/local/cuda/bin/nvcc'

kentang-mit, 2022-09-17:
Hello @BHC1205,

Would you mind checking where your `nvcc` compiler is in your OS? Perhaps you will need to add this directory to the `PATH`.

Best,
Haotian

xxlbigbrother, 2022-09-23:
Hi, I meet a similar problem and I solve it. 
problem: error: command '/usr/local/cuda/bin/nvcc' failed with exit code 1
solution:
[image]

kentang-mit, 2022-09-24:
Thank you @xxlbigbrother for the comment!

BHC1205, 2022-09-26:
Hi,I finally solved the problem.
This is an issue of lack of permissions, But that didn't happen on another machine in the same environment.

kentang-mit, 2022-09-26:
Glad to hear that. I will temporarily close the issue since it is resolved.

closed 2022-09-26 by kentang-mit