mit-han-lab/bevfusion #139: How to split the dataset into only Sunny/Rainy/Day/Night scene?

RoyZry98, 2022-09-20:
I notice that you conduct experiments with only Sunny/Rainy/Day/Night scenes in your work. I want to know how you split the dataset into those 4 scenes. Thanks a lot!

ssuralcmu, 2022-09-21:
It might be possible to use the description information in the list_scenes function in nuscenes-devkit to get keywords such as night and rain to classify the scenes. Here is a useful Colab tutorial for the same- https://colab.research.google.com/github/nutonomy/nuscenes-devkit/blob/master/python-sdk/tutorials/nuscenes_tutorial.ipynb

kentang-mit, 2022-09-21:
Thanks @ssuralcmu. These experiments are done by @Hanyuezhuohua, perhaps he can provide more insights.

Hanyuezhuohua, 2022-09-21:
As @ssuralcmu mentions, we manually split these scenes following the scene description in nuscenes dataset(the path is data/nuscenes/v1.0-trainval/scene.json). The Rainy/Night scenes are explicitly given in the "descripton". After that, it is easy to find the Sunny/Rainy/Day/Night val scenes in its original split: [](https://github.com/nutonomy/nuscenes-devkit/blob/master/python-sdk/nuscenes/utils/splits.py#L102). Hope it will be helpful to you！

kentang-mit, 2022-09-28:
Temporarily closed due to inactivity.

closed 2022-09-28 by kentang-mit