mit-han-lab/bevfusion #140: Detection results on nuscenes train/val/test splits

dsx0511, 2022-09-20:
Dear authors,

thank you very much for your amazing work. 

I want to kindly ask if it is possible to provide the detection results on nuscenes train, validation and test splits, similar to the one from Centerpoint: https://github.com/tianweiy/CenterPoint/issues/249? It would be very helpful for the people who are working on 3D multi object tracking via data association. Also, public detections help the MOT community to have a fair evaluation and comparison. Thanks a lot!

Best,
Shuxiao

kentang-mit, 2022-09-21:
Thanks for your interest. We received multiple requests on this topic, and our current suggestion is to use our predictions on the validation set (which can be obtained using this open source repo) to verify the effectiveness of your tracking algorithm. As an example, the current state-of-the-art solution, [CAMO-MOT](https://arxiv.org/pdf/2209.02540.pdf), did that on nuScenes.

kentang-mit, 2022-09-26:
Temporarily closed due to inactivity.

closed 2022-09-26 by kentang-mit