mit-han-lab/bevfusion #143: Test questions

zhenghw111, 2022-09-23:
When I execute the "torchpack dist-run -np 8 python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth --eval bbox" command,it stops at the following position:
[image]

kentang-mit, 2022-09-24:
Would you mind trying out single GPU evaluation (i.e. `-np 1`) or adding the `-v` flag to your command? Besides, there are two on-going PRs working on docker support for BEVFusion. I would also recommend you to have a try at them:

[PR](https://github.com/mit-han-lab/bevfusion/pull/144) from @bentherien and [PR](https://github.com/mit-han-lab/bevfusion/pull/145) from me.

zhenghw111, 2022-09-24:
Thank! This problem has been solved.
Can you tell me the training command?

kentang-mit, 2022-09-24:
@zhenghw111, the training code is under internal review and quality check, we will keep you posted on the progress of code release.

zhenghw111, 2022-09-24:
OK, thank you!

kkangshen, 2022-09-26:
作者你好，请问模型架构中fuser层的作用是什么？

kkangshen, 2022-09-26:
@kentang-mit  作者你好，请问模型架构中fuser层的作用是什么？

kkangshen, 2022-09-26:
@kentang-mit @zhenghw111 请问在bevfusion.py中，camera得到的future的维度为[1,80,128,128]，lidar得到的future的维度为[1，256，180，180]，他们的维度最后两维不相等，这是为什么呢？
[image]

kentang-mit, 2022-09-26:
Hi @kkangshen,

I will first translate your questions into English such that other readers might understand the context better.

1. What's the function of the `fuser` in BEVFusion?

This module is used to combine the information from camera and LiDAR sensors. In our experiments, we confirm that it is sufficient to use only one convolution layer (followed by BN/ReLU).

2. Mismatched dimensionalities: the output of the camera encoder is in [1, 80, 128, 128] and the output of the LiDAR encoder is [1, 256, 180, 180]. 

I think I kept output resolution of both branches the same. For detection it is [180, 180], and for segmentation it is [128, 128]. 

Best,
Haotian

kkangshen, 2022-09-27:
Hi @kentang-mit 
Thanks for your answer, but there are still some problems here, if the two dimensions are not equal the output of the camera encoder is in [1, 80, 128, 128] and the output of the LiDAR encoder is [1, 256, 180, 180 ], there will be an error when **torch.cat(inputs, dim=1)** how should this problem be solved.
[image]

kentang-mit, 2022-09-27:
Hi @kkangshen,

I would suggest you to use the same output resolutions for both branches. 

If you really want to have different resolutions for camera and LiDAR branches, I would suggest you to have a look at [this class](https://github.com/mit-han-lab/bevfusion/blob/2bf96604feab90edd18591a43bee1b9c41c26002/mmdet3d/models/heads/segm/vanilla.py#L47) in our segmentation head implementation, which actually changes the resolution of the output BEV feature map to a target size. You can use a similar resolution to align the resolutions of both branches first and then concatenate them together for further processing.

Best,
Haotian

kkangshen, 2022-09-30:
s

kkangshen, 2022-09-30:
Hi @kentang-mit 
Thank you very much for replying，I have the following questions。
1. I run the test.py file torchpack dist-run -np 8 python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion- det.pth --eval bbox,
When running, it is found that the dimension obtained by extract_camera_features is different from the dimension obtained by extract_lidar_features. If I want to make the two feature dimensions the same, how should I modify it.
2. I want to do the detection task, do I need to convert the features obtained by extract_camera_features into [1, 80, 180, 180]
3. When I passed the debug script, I found that when the checkpoint = load_checkpoint(model, args.checkpoint, map_location="cpu") was executed, the dx, nx, and bx parameters in the configuration file were changed. I don't know if this caused the extract_camera_features The feature becomes [1, 80, 128, 128]

kentang-mit, 2022-10-03:
That sounds pretty interesting to me @kkangshen. I can confirm that your observation in 3 is reasonable because `dx`, `nx` and `bx` are defined as model parameters, so loading the checkpoint might override these values in the configuration file. But I actually have tested these pretrained models we released and I did not run into this issue. Would you mind downloading the latest checkpoints again (because we updated them very recently) and make sure that the correct checkpoint is loaded (for example, please do not load the segmentation checkpoint if you want to do detection)?

zhenghw111, 2022-10-05:
@kkangshen 请问在pycharm上调试时是如何启动torchpack库的？我现在无法调试...

kentang-mit, 2022-10-05:
The problem related to PyCharm debugging has been resolved in another issue. I'm closing this one since it is resolved.

closed 2022-10-05 by kentang-mit

kkangshen, 2022-10-06:
> @kkangshen 请问在pycharm上调试时是如何启动torchpack库的？我现在无法调试...
你将dist.init()注释掉 ，使用单个gpu就可以

lilkeker, 2022-10-09:
@kkangshen I also want to debug on pycharm now. I can run the program in the terminal, but the debug parameter is passed in and an error is reported.
[image]
When debugging, I commented out dist.init() as you said, and set distributed to False,
[image]
I think my dataloader is fine, because the terminal input command can run the program,do you know why？please