mit-han-lab/bevfusion #147: MPI_INIT failed for some reason

lilkeker, 2022-09-26:
[image]
I run
torchpack dist-run -np 8 python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth --eval bbox
But I encountered a problem, I did install openmpi4.0.4 and mpi4py3.0.3

kentang-mit, 2022-09-26:
Hi @lilkeker,

Would you mind having a look at the [dev/docker](https://github.com/mit-han-lab/bevfusion/tree/dev/docker) branch and see whether it is helpful?

Best,
Haotian

lilkeker, 2022-09-28:
Hi Haotian, thanks for your reply, I think I have solved the MPI_INIT problem,
But now I have encountered a new problem, is there some problem in my data set?  @kentang-mit 
[image]
I noticed that the pkl file in the nuscenes dataset does not contain info["location"] information,
[image]
Here is the distribution of my dataset，some of these folders are generated by code I'm running from other projects
[image]

kentang-mit, 2022-09-29:
I see, if you are using `mmdet3d` from other projects, perhaps you can modify `camera_info["camera_intrinsics"]` to `camera_info["cam_intrinsic"]` to see whether it works.