mit-han-lab/bevfusion #148: Test question about spconv

zhangdi-trying, 2022-09-26:
I run the code "torchpack dist-run -np 1 -v python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth --eval bbox
"
but it reminds me with "Traceback (most recent call last):
  File "tools/test.py", line 230, in <module>
    main()
  File "tools/test.py", line 203, in main
    outputs = multi_gpu_test(model, data_loader, args.tmpdir, args.gpu_collect)
  File "/home/zhangdi/miniconda3/envs/bevfusion/lib/python3.8/site-packages/mmdet/apis/test.py", line 109, in multi_gpu_test
    result = model(return_loss=False, rescale=True, **data)
  File "/home/zhangdi/miniconda3/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/zhangdi/miniconda3/envs/bevfusion/lib/python3.8/site-packages/torch/nn/parallel/distributed.py", line 886, in forward
    output = self.module(*inputs[0], **kwargs[0])
  File "/home/zhangdi/miniconda3/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/zhangdi/miniconda3/envs/bevfusion/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/zhangdi/project/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 182, in forward
    feature = self.extract_lidar_features(points)
  File "/home/zhangdi/project/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 125, in extract_lidar_features
    x = self.encoders["lidar"]["backbone"](feats, coords, batch_size, sizes=sizes)
  File "/home/zhangdi/miniconda3/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/zhangdi/miniconda3/envs/bevfusion/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/zhangdi/project/bevfusion/mmdet3d/models/backbones/sparse_encoder.py", line 116, in forward
    x = self.conv_input(input_sp_tensor)
  File "/home/zhangdi/miniconda3/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/zhangdi/project/bevfusion/mmdet3d/ops/spconv/modules.py", line 132, in forward
    input = module(input)
  File "/home/zhangdi/miniconda3/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/zhangdi/project/bevfusion/mmdet3d/ops/spconv/conv.py", line 163, in forward
    outids, indice_pairs, indice_pair_num = ops.get_indice_pairs(
  File "/home/zhangdi/project/bevfusion/mmdet3d/ops/spconv/ops.py", line 92, in get_indice_pairs
    return get_indice_pairs_func(
RuntimeError: /home/zhangdi/project/bevfusion/mmdet3d/ops/spconv/src/indice_cuda.cu 124
cuda execution failed with error 2
"

Map be there exists some error with the spconv? But import spconv is ok.
[image]

kentang-mit, 2022-09-26:
Hi @zhangdi-trying,

The current version is based on spconv 1.2.1. You probably need to use the backend under `mmdet3d/ops/spconv`. Support for spconv 2.x has been tested internally and will be pushed into the main branch in the future.

Best,
Haotian

zhangdi-trying, 2022-09-27:
>

> 

Hello, I just contact the mmdect and I don't know how to use the backend under mmdet3d/ops/spconv. Could you tell me about that ? Thanks a lot.

kentang-mit, 2022-09-27:
Hi @zhangdi-trying,

You can just run

```bash
python setup.py develop
```

under the bevfusion project folder. I would recommend you to start a new environment without spconv 2.x installed to avoid conflicts.

Best,
Haotian

Jackjie8, 2022-09-28:
I also have the same problem. I followed the last comment to run setup.py, but it didn't work. Then I installed  spconv-1.2.1, but the error  occurred again. 
Could you tell me how to solve it ?

zhangdi-trying, 2022-09-29:
> Hi @zhangdi-trying,
> 
> The current version is based on spconv 1.2.1. You probably need to use the backend under `mmdet3d/ops/spconv`. Support for spconv 2.x has been tested internally and will be pushed into the main branch in the future.
> 
> Best, Haotian

I have tried that but
[image]
cannot import spconv

jiajiaen, 2022-09-29:
Hello, I just contact the mmdect and I don't know how to use the backend under mmdet3d/ops/spconv. Could you tell me about that ? Thanks a lot.

kentang-mit, 2022-09-29:
Hi @Jackjie8, as I have mentioned previously, we made custom modifications on `spconv-1.2.1`, so it might not be the best idea to install the official `spconv` library.  

Hi @zhangdi-trying and @jiajiaen, if you have difficulties in setting up the environment, I would suggest you guys to take a look at our instructions on docker in README.md. Hopefully that will be helpful. `spconv` should be installed as part of the entire `mmdet3d/ops` directory during the setup develop process.

serend1p1ty, 2022-09-30:
Hello, @kentang-mit, thanks for your great work. I also met this problem.

`bevfusion` now use `spconv==1.2.1` (if we follow the installation guidance, run `python setup.py develop`).

In other issues, you have mentioned switching to `spconv==2.x` can avoid this problem. How can we switch to 2.x?

Jackjie8, 2022-10-02:
@kentang-mit Hello, thanks for your response. As you said, the setup.py will install the environment, but the errors about indice_cuda.cu in spconv still exist.
[image]

serend1p1ty, 2022-10-02:
> @kentang-mit Hello, thanks for your response. As you said, the setup.py will install the environment, but the errors about indice_cuda.cu in spconv still exist. [image]

Exactly the same problem.

kentang-mit, 2022-10-03:
Hi @serend1p1ty,

For spconv 2.x support, we will release it in the future because we have limited bandwidth recently. 

Hi @serend1p1ty and @Jackjie8,

What about trying out our docker setup first?

Best,
Haotian

zhangdi-trying, 2022-10-04:
> Hi @serend1p1ty,
> 
> For spconv 2.x support, we will release it in the future because we have limited bandwidth recently.
> 
> Hi @serend1p1ty and @Jackjie8,
> 
> What about trying out our docker setup first?
> 
> Best, Haotian

Erros again. I try to use the docker setup But when I run "docker build . -t bevfusion", It reminds me with
[image]
How can I solve this? Thanks a lot!

kentang-mit, 2022-10-04:
I think I did not specify a mirror from Tsinghua University. Would you mind trying out downloading from the official Ubuntu mirror site?

zhangdi-trying, 2022-10-05:
> I don't think I used a mirror from Tsinghua University. Would you mind trying out the official Ubuntu mirror site? If that's too slow, definitely have a look at other mirror sites in China, for example, the one from USTC.

Thanks for your great work and patient answer, I have solved this problems! Thanks again

serend1p1ty, 2022-10-05:
Hi, @zhangdi-trying

Can the Docker environment solve the problem of spconv?

zhangdi-trying, 2022-10-05:
> 
[image]

I have tried but fail again

kentang-mit, 2022-10-05:
Hi @serend1p1ty,

If you follow our guidance in README.md I think it should solve the problem of spconv. Because you will have identical setup as mine.

Hi @zhangdi-trying,

Did you also run 
```bash
python setup develop
``` 
in the docker environment?

Best,
Haotian

zhangdi-trying, 2022-10-05:
> Hi @serend1p1ty,
> 
> If you follow our guidance in README.md I think it should solve the problem of spconv. Because you will have identical setup as mine.
> 
> Hi @zhangdi-trying,
> 
> Did you also run
> 
> ```shell
> python setup develop
> ```
> 
> in the docker environment?
> 
> Best, Haotian

I have run that code before. And in case some errors, I recreated a container again, but still failed. I don't know the reason.
[image]

kentang-mit, 2022-10-06:
Just to clarify, did you run that code inside the docker? If not, please do so.

BII-wushuang, 2022-10-07:
My experience is that this spconv runtime error occurs if the CUDA driver on your machine is not at least 11.3.
```
RuntimeError: /bevfusion/mmdet3d/ops/spconv/src/indice_cuda.cu 124
cuda execution failed with error 2
```

zhangdi-trying, 2022-10-07:
> My experience is that this spconv runtime error occurs if the CUDA toolkit on your machine is not at least 11.3. Otherwise, the Docker environment should work perfectly.
> 
> ```
> RuntimeError: /bevfusion/mmdet3d/ops/spconv/src/indice_cuda.cu 124
> cuda execution failed with error 2
> ```

My cuda version is 11.3, but there still exists some error. I will try on another machine.

zhangdi-trying, 2022-10-07:
> Hi, @zhangdi-trying
> 
> Can the Docker environment solve the problem of spconv?

I couldn't, and you?

BII-wushuang, 2022-10-07:
Another hack I tried that works is replacing all `4096` in `mmdet3d/ops/spconv/src/indice_cuda.cu` with `256` and then 
```
python setup.py install
```
This is on a machine with CUDA 11.1 toolkit and the author's Docker environment.

BII-wushuang, 2022-10-07:
Another hack I tried that works (on a machine with CUDA 11.1 toolkit and the authors' Docker environment) is replacing all `4096` in `mmdet3d/ops/spconv/src/indice_cuda.cu` with `256` and then 
```
python setup.py develop
```
It seems that the inference outputs are not affected by this hack.

kentang-mit, 2022-10-08:
Sounds like good comments. Thanks @BII-wushuang !

Jackjie8, 2022-10-09:
Hi, @kentang-mit 
I followed the docker setup, but another error came out when I ran the test command. Do you have any idea? Thanks!
[image]

Z-chocking, 2022-10-10:
> Another hack I tried that works (on a machine with CUDA 11.1 toolkit and the authors' Docker environment) is replacing all `4096` in `mmdet3d/ops/spconv/src/indice_cuda.cu` with `256` and then
> 
> ```
> python setup.py develop
> ```
> 
> It seems that the inference outputs are not affected by this hack.

I had followed your instruction and it works.my CUDA version is 11.3. THANK YOU!

kentang-mit, 2022-10-10:
Sounds good. I'll close this issue for now since it is resolved. Later users please use CUDA 11.3 as suggested.

closed 2022-10-10 by kentang-mit

MakiNaruto, 2022-10-25:
> Another hack I tried that works (on a machine with CUDA 11.1 toolkit and the authors' Docker environment) is replacing all `4096` in `mmdet3d/ops/spconv/src/indice_cuda.cu` with `256` and then
> 
> ```
> python setup.py develop
> ```
> 
> It seems that the inference outputs are not affected by this hack.

hi @BII-wushuang、@Z-chocking 
I use the authors' Docker environment，and I had followed your instruction and  replacing all `4096` in `mmdet3d/ops/spconv/src/indice_cuda.cu` with `256` ，and python setup.py develop，but it not work，
but I met this problem:

Traceback (most recent call last):
  File "tools/train.py", line 87, in <module>
    main()
  File "tools/train.py", line 76, in main
    train_model(
  File "/home/bevfusion/bevfusion-main/mmdet3d/apis/train.py", line 126, in train_model
    runner.run(data_loaders, [("train", 1)])
  File "/opt/conda/lib/python3.8/site-packages/mmcv/runner/epoch_based_runner.py", line 127, in run
    epoch_runner(data_loaders[i], **kwargs)
  File "/home/bevfusion/bevfusion-main/mmdet3d/runner/epoch_based_runner.py", line 14, in train
    super().train(data_loader, **kwargs)
  File "/opt/conda/lib/python3.8/site-packages/mmcv/runner/epoch_based_runner.py", line 50, in train
    self.run_iter(data_batch, train_mode=True, **kwargs)
  File "/opt/conda/lib/python3.8/site-packages/mmcv/runner/epoch_based_runner.py", line 29, in run_iter
    outputs = self.model.train_step(data_batch, self.optimizer,
  File "/opt/conda/lib/python3.8/site-packages/mmcv/parallel/distributed.py", line 52, in train_step
    output = self.module.train_step(*inputs[0], **kwargs[0])
  File "/home/bevfusion/bevfusion-main/mmdet3d/models/fusion_models/base.py", line 78, in train_step
    losses = self(**data)
  File "/opt/conda/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/opt/conda/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/bevfusion/bevfusion-main/mmdet3d/models/fusion_models/bevfusion.py", line 187, in forward
    outputs = self.forward_single(
  File "/opt/conda/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/bevfusion/bevfusion-main/mmdet3d/models/fusion_models/bevfusion.py", line 256, in forward_single
    assert len(features) == 1, features
AssertionError: [tensor([[[[0.0000, 0.0000, 0.0000,  ..., 0.0000, 0.0000, 0.0000],
          [0.0000, 0.0000, 0.0000,  ..., 0.0000, 0.0000, 0.0000],
          [0.0000, 0.0000, 0.0000,  ..., 0.0000, 0.0000, 0.0000],
          ......

Z-chocking, 2022-10-25:
> > Another hack I tried that works (on a machine with CUDA 11.1 toolkit and the authors' Docker environment) is replacing all `4096` in `mmdet3d/ops/spconv/src/indice_cuda.cu` with `256` and then
> > ```
> > python setup.py develop
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > It seems that the inference outputs are not affected by this hack.
> 
> hi @BII-wushuang、@Z-chocking I use the authors' Docker environment，and I had followed your instruction and replacing all `4096` in `mmdet3d/ops/spconv/src/indice_cuda.cu` with `256` ，and `python setup.py develop` on NVIDIA 3090, but it not work， and I met the new problem:
> 
> ```
> Traceback (most recent call last):
>   File "tools/train.py", line 87, in <module>
>     main()
>   File "tools/train.py", line 76, in main
>     train_model(
>   File "/home/bevfusion/bevfusion-main/mmdet3d/apis/train.py", line 126, in train_model
>     runner.run(data_loaders, [("train", 1)])
>   File "/opt/conda/lib/python3.8/site-packages/mmcv/runner/epoch_based_runner.py", line 127, in run
>     epoch_runner(data_loaders[i], **kwargs)
>   File "/home/bevfusion/bevfusion-main/mmdet3d/runner/epoch_based_runner.py", line 14, in train
>     super().train(data_loader, **kwargs)
>   File "/opt/conda/lib/python3.8/site-packages/mmcv/runner/epoch_based_runner.py", line 50, in train
>     self.run_iter(data_batch, train_mode=True, **kwargs)
>   File "/opt/conda/lib/python3.8/site-packages/mmcv/runner/epoch_based_runner.py", line 29, in run_iter
>     outputs = self.model.train_step(data_batch, self.optimizer,
>   File "/opt/conda/lib/python3.8/site-packages/mmcv/parallel/distributed.py", line 52, in train_step
>     output = self.module.train_step(*inputs[0], **kwargs[0])
>   File "/home/bevfusion/bevfusion-main/mmdet3d/models/fusion_models/base.py", line 78, in train_step
>     losses = self(**data)
>   File "/opt/conda/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
>     return forward_call(*input, **kwargs)
>   File "/opt/conda/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
>     output = old_func(*new_args, **new_kwargs)
>   File "/home/bevfusion/bevfusion-main/mmdet3d/models/fusion_models/bevfusion.py", line 187, in forward
>     outputs = self.forward_single(
>   File "/opt/conda/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
>     output = old_func(*new_args, **new_kwargs)
>   File "/home/bevfusion/bevfusion-main/mmdet3d/models/fusion_models/bevfusion.py", line 256, in forward_single
>     assert len(features) == 1, features
> AssertionError: [tensor([[[[0.0000, 0.0000, 0.0000,  ..., 0.0000, 0.0000, 0.0000],
>           [0.0000, 0.0000, 0.0000,  ..., 0.0000, 0.0000, 0.0000],
>           [0.0000, 0.0000, 0.0000,  ..., 0.0000, 0.0000, 0.0000],
>           ......
> ```

Sorry i have not met this error before,but i builded this project without docker.

MakiNaruto, 2022-10-26:
> Sorry i have not met this error before,but i builded this project without docker.

excuse me，is that u use 3090 Graphic card？ or 2080Ti ?

Z-chocking, 2022-10-26:
> > Sorry i have not met this error before,but i builded this project without docker.
> 
> excuse me，is that u use 3090 Graphic card？ or 2080Ti ?

I use RTX 3090

MakiNaruto, 2022-10-26:
> I use RTX 3090
Do you mind if I contact you by email? Maybe it's more convenient。My email is become007@icloud.com

hitbuyi, 2024-04-18:
> > 
> 
> [image]
> 
> I have tried but fail again

I have the same problem, the mmdet3d doesn't install udner  python's site-packages folder