mit-han-lab/bevfusion #149: Problem about image preprocessing

HatakeKiki, 2022-09-26:
Hi! Thank you for your great work first! 
I noticed the preprocess of image before passing them to camera backbone has two steps:
  1.   original images at size of 1600*900 are scaled into 768*432 (by multiplying 0.48)
  2.   random crop are applied and the image size is 704*256
The random crop may cause almost half of the pictures unused. Some ground truth objects may have no features from camera at all, hence no features for some objects in camera only method and impossible to be detected. And I didn't notice any related label filter?
Wouldn't this lead to poor performance when random crop happens to crop majority of the the objects on the image?

HatakeKiki, 2022-09-26:
Also the random crop will lead to different frustums in 3D world since projection is different and the precomputation part is not feasible in this way?

kentang-mit, 2022-09-26:
1. I think your understanding is correct. The upper part of the pictures is not used. There are two reasons why this might not be a large issue in terms of accuracy:

- First, BEVFusion utilizes both camera and LiDAR inputs. The LiDAR branch has full information and we view the camera branch as a source of auxiliary information.
- Second, if you visualize the samples on nuScenes, you will find that the upper part of the images usually does not contain foreground objects. 

2. Regarding precomputation, it happens only in the test phase. Since we apply a fixed cropping strategy (keep the bottom 256 pixels of the images), the projection matrices will still be the same after cropping if they are the same before such operation.

HatakeKiki, 2022-09-26:
So it's a fixed crop instead of random crop. I got this wrong. Thank you for your replay and looking forward to the release of training details.

kentang-mit, 2022-09-26:
Thanks. Our training code is currently available [here](https://github.com/mit-han-lab/bevfusion/tree/dev/training). It is under review and I expect it to be merged very soon.

closed 2022-09-26 by kentang-mit

HatakeKiki, 2022-09-26:
Ohhhhhh! Thank you so much! Now I can try reproduction~

SeungHwi0613, 2023-06-15:
> Hi! Thank you for your great work first! I noticed the preprocess of image before passing them to camera backbone has two steps:
>
> 1. original images at size of 1600_900 are scaled into 768_432 (by [...]

Hello! where can i check about scaling into 768432?