mit-han-lab/bevfusion #153: torchpack version?

HengLoveYi, 2022-09-27:
Hello,
        I would like to ask whick torchpack version do you use? I use version 0.3.1 but it failed to import 'torchpack.utils.tqdm
' when I run visualize.py.

 root@LGSYS-AP01:/DATA_YJY/DATA_DIR/LiJiaheng/code/bevfusion# python tools/visualize.py 
Traceback (most recent call last):
  File "tools/visualize.py", line 13, in <module>
    from torchpack.utils.tqdm import tqdm
 
      Thank you for your excellent work and expect for your reply!

Best regards!

VeeranjaneyuluToka, 2022-09-27:
Doing this change should work straight away

#from torchpack.utils.tqdm import tqdm
from tqdm import tqdm

HengLoveYi, 2022-09-28:
Hello, I have try it but it still has a problem, as bellow:

Traceback (most recent call last):
  File "tools/visualize.py", line 173, in <module>
    main()
  File "tools/visualize.py", line 95, in main
    outputs = model(*data)
  File "/home/yjy/anaconda3/envs/Lee02/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/yjy/anaconda3/envs/Lee02/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 98, in new_func
    return old_func(*args, **kwargs)
  File "/DATA_YJY/DATA_DIR/LiJiaheng/code/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 174, in forward
    feature = self.extract_camera_features(
  File "/DATA_YJY/DATA_DIR/LiJiaheng/code/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 101, in extract_camera_features
    B, N, C, H, W = x.size()
AttributeError: 'str' object has no attribute 'size'

  It seems data type problem. Before run visualize.py, I have run test.py and succeeded, so the environment should be correct.

  Expect for your reply.

HengLoveYi, 2022-09-28:
Thank you for your reply very much and it works !!!

closed 2022-09-30 by zhijian-liu