mit-han-lab/bevfusion #154: changes for Monocular Image data

RG2806, 2022-09-27:
Hey,
      Thanks for open sourcing this awesome work.
       I wanted to train your C+L  fusion model on my own dataset but i have a monocular camera, thus only one image view. Can you point me to what changes i would have to make to train the model successfully.

kentang-mit, 2022-09-28:
Hi @RG2806,

I think BEVFusion will still work with monocular camera + LiDAR inputs. The main difference between single camera v.s. surrounding camera is that the `img` variable will be 1x3x256x704 instead of 6x3x256x704, `lidar2camera` will be something like 1x3x3 instead of 6x3x3, similar for other variables. 

Best,
Haotian

RG2806, 2022-09-28:
Hi @kentang-mit 
Thanks, also the size of nuscenes images is 1600 × 1200.
Then how do you arive at 256×704?

My dataset has images of size 4096×3000. Apart from default.yml. how would i use it?

kentang-mit, 2022-09-29:
I'm just using these numbers as examples. You can have any resolution you want. Simply changing `image_size` and `augment_2d` parameters in [this file](https://github.com/mit-han-lab/bevfusion/blob/main/configs/nuscenes/default.yaml) will be a good idea. 

Note that [this line](https://github.com/mit-han-lab/bevfusion/blob/2bf96604feab90edd18591a43bee1b9c41c26002/configs/nuscenes/default.yaml#L14) means: the training time resolution is sampled in the range of 0.38x to 0.55x of the original resolution, and the test time resolution scaler is sampled from the range of 0.48x to 0.48x of the original resolution.

RG2806, 2022-09-29:
Thanks for the clarification

closed 2022-09-29 by RG2806