mit-han-lab/bevfusion #155: Dataset without velocity input

RG2806, 2022-09-27:
Hi,
Thanks for the great work. I wanted to run your model training on my custom dataset, but i don't have the velocity field in my ground truth. I created custom dataset.py file for it. The code errors out at IoU calculation at
assert bboxes1.size(-1) == bboxes2.size() >=7
bboxes1, i.e, prediction is of shape (n,9) and
bboxes2 ,i.e, GT is of shape (m,7)
Can you guide for the changes to be made in the training code to tackle this?

kentang-mit, 2022-09-28:
In this case you can pad dummy velocity (e.g. [0, 0] for each object) to the GT boxes. Besides, in loss calculation you can assign zero weights to the dimensions corresponding to the velocity.

RG2806, 2022-09-28:
Hi,
I am training a fusion model with the 
config: configs/dataset/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml 

for changing the weights to zero, do you mean at this 
https://github.com/mit-han-lab/bevfusion/blob/main/mmdet3d/models/heads/bbox/transfusion.py#L391
if so:
       the bbox targets and weights have dimensions have size (4*200*10); btw i have 4 labels in that frame.
what should i change?

kentang-mit, 2022-09-29:
You can just change the last two dimensions [here](https://github.com/mit-han-lab/bevfusion/blob/2bf96604feab90edd18591a43bee1b9c41c26002/configs/nuscenes/det/transfusion/default.yaml#L26) from 0.2 to 0.

RG2806, 2022-09-29:
Hey @kentang-mit ,
thanks, i'll change accordingly.
Can you tell what these weights are for like,
3 for centre
3 for dims
1 for yaw
2 for velocity
and then there is 1 more?

kentang-mit, 2022-09-30:
There are two dimensions for yaw corresponding to sin and cos of the angle.

RG2806, 2022-09-30:
Thanks for the clarification.

closed 2022-09-30 by RG2806