mit-han-lab/bevfusion #157: Segmentation fault(core dumped)

jiajiaen, 2022-09-29:
Hello, when I run: "torchpack dist-run -np 2 python tools/train.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml --model.encoders.camera.backbone.init_cfg.checkpoint pretrained/swint-nuimages-pretrained.pth" to train the model,
I get an error:

 *** Process received signal ***
 Signal: Segmentation fault (11)
 Signal code: Address not mapped (1)
Failing at address: 0xfffffee8

This error occurs when the model starts training or after training for a while, I tried to run: python setup.py develop  to recompile the code, but the problem is still not solved, I suspect it is the version of spconv. Please how can I solve this problem, thank you.

kentang-mit, 2022-09-29:
What about trying out the docker? We have provided some instructions in the README.md.

jiajiaen, 2022-09-30:
> 

Thanks for your reply, I will try to use docker to configure the environment, when will the training method of the fusion part be released. Thanks again for your great work.

kentang-mit, 2022-10-04:
Temporarily closed since the issue is resolved.

closed 2022-10-04 by kentang-mit