mit-han-lab/bevfusion #159: training time on RTX3090

EhsanRusta, 2022-10-02:
I just wanna know how long takes time to train on RTX3090 GPU? 
I'm using Nuscence training dataset. appreciate if you advise me how to train ...

kentang-mit, 2022-10-03:
Training commands are available in our README.md. I haven't estimated the training time on RTX3090, but on A6000 (with similar computation capability) it is less than 1 day for single modality models and fusion models will need approximately 10 hours more. We assume 8 GPUs are used.

closed 2022-10-03 by kentang-mit

josh-wende, 2024-01-24:
Do you mean that the total training time for a fusion model would be ~24hrs for the lidar-only model, plus 10hrs more for ~34 total? Or ~24hrs for the lidar-only model plus _another_ ~24+10hrs for the fusion model?