mit-han-lab/bevfusion #168: About image rotation augmentation

YurongYou, 2022-10-05:
Hi Zhijian and Haotian,

Thanks for open sourcing the BEVFusion! It is a great work and I like it a lot!

I was reading through the code and I am a bit confused by the [image augmentation code](https://github.com/mit-han-lab/bevfusion/blob/2bf96604feab90edd18591a43bee1b9c41c26002/mmdet3d/datasets/pipelines/transforms_3d.py#L65-L90): Are there specific reasons you do the rotation at the last step? It seems that doing so will result in worse image quality (jagged pixels):
```python
img = Image.open("./data/nuscenes/samples/CAM_FRONT/n015-2018-07-11-11-54-16+0800__CAM_FRONT__1531281560162460.jpg")
transform_image = img.resize(resize_dims)
transform_image = transform_image.crop(crop)
transform_image = transform_image.rotate(rotate)
```
[image]

compared with rotate first:
```python
img = Image.open("./data/nuscenes/samples/CAM_FRONT/n015-2018-07-11-11-54-16+0800__CAM_FRONT__1531281560162460.jpg")
transform_image = img.rotate(rotate)
transform_image = transform_image.resize(resize_dims)
transform_image = transform_image.crop(crop)
```
[image]

Also, I am guessing the [flip augmentation matrix here](https://github.com/mit-han-lab/bevfusion/blob/2bf96604feab90edd18591a43bee1b9c41c26002/mmdet3d/datasets/pipelines/transforms_3d.py#L77) should probably be
```python
b = torch.Tensor([crop[2] - crop[0] - 1, 0])
```
according to [PIL flipping code](https://github.com/python-pillow/Pillow/blob/95cff6e959bb3c37848158ed2145d49d49806a31/src/libImaging/Geometry.c#L31-L39).

Thanks again for your great work!

kentang-mit, 2022-10-06:
Hi @YurongYou,

Thanks for the comments! For the image rotation augmentation, I believe we just follow the original implementation of LSS when we implement it. For flip augmentation, it is a good point, but I guess the difference is not very large. How do you think about it @zhijian-liu ?

Best,
Haotian

zhijian-liu, 2022-10-06:
@kentang-mit, I think we should make it correct. Could you please re-run some experiments when you have the bandwidth?

kentang-mit, 2022-10-06:
@zhijian-liu, sure, I will have a try recently, perhaps late this week or early next week.

YurongYou, 2022-10-06:
Thanks! For the rotation-first implementation, I just implemented the corresponding post-homography transformation (need to input `ori_shape`). Feel free to double-check and use it :)
```python
# adjust image
img = img.rotate(rotate)
img = img.resize(resize_dims)
img = img.crop(crop)
if flip:
    img = img.transpose(method=Image.FLIP_LEFT_RIGHT)
# post-homography transformation
W, H = ori_shape
theta = rotate / 180 * np.pi
A = torch.Tensor(
    [
        [np.cos(theta), np.sin(theta)],
        [-np.sin(theta), np.cos(theta)],
    ]
)
b = torch.Tensor([W, H]) / 2
b = A.matmul(-b) + b

rotation = A.matmul(rotation)
translation = A.matmul(translation) + b

rotation *= resize
translation *= resize
translation -= torch.Tensor(crop[:2])
if flip:
    A = torch.Tensor([[-1, 0], [0, 1]])
    b = torch.Tensor([crop[2] - crop[0] - 1, 0])
    rotation = A.matmul(rotation)
    translation = A.matmul(translation) + b
```

kentang-mit, 2022-10-06:
Thanks a lot!

VeeranjaneyuluToka, 2023-04-18:
Hi, When i dump image tensor after image transform, it shows as below.
[image]

I believe there is something wrong here, is not it? if it is so, could you tell me what is going wrong here? Thanks!

lcc815, 2023-05-04:
Could any one explain why should we do that in this way? 
I am a quite junior student and my own dataset's coordinate system is different from that of nuscenes. More specificly, the x-axis points to the front of the vehicle, and the y-axis points to the left of the vehicle, while for nuscese x-axis to right, y-axis to front.
In the above mentioned cases, should I change the code?

lcc815, 2023-05-05:
After learning for a day, I seem understand the reason.
However, I think the [transform matrix here](https://github.com/mit-han-lab/bevfusion/blob/main/mmdet3d/datasets/pipelines/transforms_3d.py#L114) should probably be
`            transform[:2, 2] = translation`

closed 2024-05-04 by zhijian-liu