mit-han-lab/bevfusion #170: Question about camera-only test

fangliancheng, 2022-10-06:
Hi,

I was trying to test the provided pre-trained camera-only model on Nuscene, specifically,  I am running the following command:

```shell
torchpack dist-run -np 2 python tools/test.py configs/nuscenes/det/centerhead/lssfpn/camera/256x704/swint/default.yaml  pretrained/camera-only-det.pth --eval bbox
``` 
The eval result I get is the following, which is much lower than your results(mAP 35.56).
mAP: 0.1726                              
mATE: 0.8025
mASE: 0.4065
mAOE: 0.6763
mAVE: 1.3682
mAAE: 0.4810
NDS: 0.2497

I am wondering is there anything I did wrong? Thanks! 

Best regards,
Liancheng

kentang-mit, 2022-10-06:
Hi @fangliancheng,

Could you please check whether you have used the latest code and checkpoints? There are some changes (in late September) to the input preprocessing and our pretrained checkpoint. As a result, if one of your [code, checkpoint] is not the latest, the evaluation metrics will not be correct.

Best,
Haotian

fangliancheng, 2022-10-07:
I cloned the latest repo again, now I can reproduce the result, thanks!