mit-han-lab/bevfusion #171: Question about tracking

whoakang, 2022-10-08:
Thanks for your amazing job! have a question that how to use centerpoint to achieve 3D box tracking？

kentang-mit, 2022-10-08:
I think we used the tracker from Tianwei's CenterPoint [implementation](https://github.com/tianweiy/CenterPoint/tree/master/tools/nusc_tracking) and fed in our detection predictions.

whoakang, 2022-10-08:
Thanks for your reply！
i test detection model used this command：torchpack dist-run -np 8 python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth --eval=bbox

should i change yaml file to this：configs/nuscenes/det/centerhead/lssfpn/camera/256x704/swint/default.yaml, to achieve tracking?

kentang-mit, 2022-10-08:
You can just save the predictions from our detector and feed it to a separate tracker module from Tianwei. I think that will be the easiest way to use BEVFusion as a tracking baseline.

whoakang, 2022-10-09:
ok，i’will try this way，thanks.

kentang-mit, 2022-10-10:
Great. I'll close this issue since it is resolved.

closed 2022-10-10 by kentang-mit