mit-han-lab/bevfusion #173: Test question

whoakang, 2022-10-08:
when i use command to test model:
torchpack dist-run -np 8 python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml
pretrained/bevfusion-det.pth --eval=bbox

the test info prints : Results writes to /tmp/tmpo54vrfop/results/results_nusc.json
how to change the save path in code?
waiting for your replay!

kentang-mit, 2022-10-08:
If you want to save the results in json format, please try out adding this flag to your test command:
```bash
--eval-options "jsonfile_prefix=[desired save path]"
```

If you want to save the results in pkl format (which is more readable), please try out adding this flag:
```bash
--out [desired path].pkl
```

whoakang, 2022-10-09:
Thanks！Have a nice day~~

kentang-mit, 2022-10-10:
Great, I will close this issue since it is resolved.

closed 2022-10-10 by kentang-mit