mit-han-lab/bevfusion #179: method for BEV encoder

roywithfiringblade, 2022-10-10:
Thanks for sharing the excellent work, I noticed in paper it says the BEV encoder is made of a few res blocks. I checked the code, that SECOND decoder and SECOND FPN are utilized. Is there a reason behind that. Can you give the performance gap between those 2 methods?
Looking forward to your reply!

kentang-mit, 2022-10-10:
Hi @roywithfiringblade,

Actually we just follow the conventional CenterPoint design in our implementation to compare our method fairly with other methods. The sentence you see in the paper actually wants to convey the following message: because we have a lot of conv layers after fusion, it is unnecessary to make the fuser complicated (for example, adding a few attention / deformable convolution layers so that the misalignment between camera and LiDAR feature maps can be fixed). Hope my explanation will clarify your questions.

Best,
Haotian

roywithfiringblade, 2022-10-10:
@kentang-mit   Thanks for your quick reply. My point is that from the config files, `SECOND` and `SECOND FPN` are used for camera+lidar fusion. (below `transfusion` directory) and for only camera input, `GeneralizedResNet`  and `LSSFPN` are used. (below `centerpoint` directory). I want to now is there any reason behind that. Thanks!

kentang-mit, 2022-10-11:
I think that is because almost all LiDAR-based methods and previous fusion methods all used SECOND+SECOND FPN (e.g.: CenterPoint, MVP, TransFusion, PointAugmenting, and a lot of others), and for all camera-based methods they use GeneralizedResNet+LSSFPN (LSS, BEVDet, and later BEVDepth, and a lot of others). We just want to make our settings aligned with other research papers.

roywithfiringblade, 2022-10-11:
Thanks for your reply!