mit-han-lab/bevfusion #181: 2D backbone of BEVFusion-base

HatakeKiki, 2022-10-12:
Hi! Congratulations on your newly submitted results on nuScenes Leaderboard!

As you mentioned in the description that you use a higher input resolution, does that mean the higher input resolution of the images? Like the one you used in ablation study, Table 4c in your paper.

I'm not so familiar with image processing but it seems that the change of input resolution will lead to the change of input channels, hence the structure of camera backbone will be changed. So did you just use a different backbone or alter the original backbone and retrained it? Could you please provide the new backbone you are using so I can better reproduce the new results?

Also, I'm a little bit confused about the sentence, "match the computation cost of concurrent work". Does it mean the updated version is more suitable for parallel computation or more compatible with other researches you're currently occupied with?

Thank you in advance.

kentang-mit, 2022-10-12:
Hi @HatakeKiki,

Thanks for your interest. Yes, we used higher input resolution, this is because the original resolution we used (704x256) is much smaller than the widely used one (800x448). As a result, the MACs of our model is in the ~250G range, while almost all other methods have at least ~500G MACs. We scale up the resolution just to match their computation budget. The backbone is not changed, it is still SwinT, the input preprocessing change we mentioned in the description has been released, you can check the commit in late September and git diff will tell you what files we changed.

Best,
Haotian

HatakeKiki, 2022-10-13:
I notice that the size of the output image after ImageAug3D is still 704*256 since the parameter image_size remains unchanged in the new config file. Unlike the previous config that resizes images from 1600*900 to 768*432 and then crops into 704*256, now it resizes images into 880*342 and then still crops into 704*256. Is this the scale-up process you mentioned?

kentang-mit, 2022-10-13:
Our input resolution is now larger than 704x256. The computation is similar to 800x448 models. But your understanding is correct, basically we downsample to a larger size and then crop.

kentang-mit, 2022-11-04:
Closed due to inactivity.

closed 2022-11-04 by kentang-mit