mit-han-lab/bevfusion #184: About reproducing the performances.

JamesHao-ml, 2022-10-13:
Thanks for releasing the awesome codes! I am trying to reproduce the performances. I have tried training convfuser.yaml using two pre-trained settings: 
1.  the released lidar-only checkpoint and image checkpoint. 
2. the provided lidar-camera model(which achieves 68.5 on val).

However, as the training epoch increases, the performance continues dropping and the best mAP I can achieve is around ~67%. Especially the second setting is actually fine-tuning the 68.5% checkpoint. So I wonder is it possible to directly reproduce the performance with the released config: https://github.com/mit-han-lab/bevfusion/blob/main/configs/nuscenes/det/transfusion/secfpn/camera%2Blidar/swint_v0p075/convfuser.yaml. If not, what should be modified? Thanks!

kentang-mit, 2022-10-13:
You will have to modify that file. Full schedule will be released in the future (my expectation is the end of next month if everything goes on well) but I believe that we have already provided enough information for people to reproduce.

closed 2022-10-13 by kentang-mit

JamesHao-ml, 2022-10-18:
@kentang-mit Appreciate your kind reply! I have been struggling in training the fusion model. Would you mind giving any hint of what modifications should I try before you release the full schedule? For example the data augmentation, lr schedule or anything else?

MakiNaruto, 2022-10-29:
@JamesHao-ml  hi, I am trying to reproduce the performances with ```torchpack dist-run 8 python tools/train.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml```  and get 62.7 mAP，and i found your question。I'm very confused about what I missed until i found your question：

> I have tried training convfuser.yaml using two pre-trained settings:
> 
> 1. the released lidar-only checkpoint and image checkpoint.
> 2. the provided lidar-camera model(which achieves 68.5 on val).

Can you tell me what command you used？how to using two pre-trained setting， look forward to your reply，thanks。

philipp-fischer, 2022-11-03:
Same thing here. We can get it trained to only 65% mAP, the released checkpoint results in 68.5% mAP. 
Would love to run the official training recipe that was used to obtain the checkpoint you published.

zlenyk, 2022-11-09:
Hi! I'm facing the same issue. Provided checkpoint prints 68.5% at the end of eval script, but I can only train up to 64mAP with the recipe. However I was training only with batch size of 16 instead of 32, that's probably why we get 1 mAP lower than @philipp-fischer.

Do you know what kind of oversampling is used on the training dataset? Number of steps per epoch suggests that we are running ~5x more steps than number of training data.
Thank you!

YangChen1234567, 2022-12-02:
hi!

> 

Have you fount how to modify the config file for reproducing the results on paper???? I'm training the fusion model, but when i load the lidar-only det model, the val map is getting lower and lower....I don‘t know why...