mit-han-lab/bevfusion #187: train error ValueError: matrix contains invalid numeric entries

skyezzz, 2022-10-15:
Hi! Thanks for sharing your excellent work!
I ran the code as follow:
`torchpack dist-run -np 2 python tools/train.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml`
It started out well, but when it came to 10 epoch, I got an error as follow:

```
Traceback (most recent call last):
  File "tools/train.py", line 87, in <module>
    main()
  File "tools/train.py", line 76, in main
    train_model(
  File "/home/seb/workspace/project/bevfusion_mit/bevfusion/mmdet3d/apis/train.py", line 126, in train_model
    runner.run(data_loaders, [("train", 1)])
  File "/home/seb/anaconda3/envs/mit_fusion/lib/python3.8/site-packages/mmcv/runner/epoch_based_runner.py", line 127, in run
    epoch_runner(data_loaders[i], **kwargs)
  File "/home/seb/workspace/project/bevfusion_mit/bevfusion/mmdet3d/runner/epoch_based_runner.py", line 14, in train
    super().train(data_loader, **kwargs)
  File "/home/seb/anaconda3/envs/mit_fusion/lib/python3.8/site-packages/mmcv/runner/epoch_based_runner.py", line 50, in train
    self.run_iter(data_batch, train_mode=True, **kwargs)
  File "/home/seb/anaconda3/envs/mit_fusion/lib/python3.8/site-packages/mmcv/runner/epoch_based_runner.py", line 29, in run_iter
    outputs = self.model.train_step(data_batch, self.optimizer,
  File "/home/seb/anaconda3/envs/mit_fusion/lib/python3.8/site-packages/mmcv/parallel/distributed.py", line 52, in train_step
    output = self.module.train_step(*inputs[0], **kwargs[0])
  File "/home/seb/workspace/project/bevfusion_mit/bevfusion/mmdet3d/models/fusion_models/base.py", line 78, in train_step
    losses = self(**data)
  File "/home/seb/anaconda3/envs/mit_fusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/seb/anaconda3/envs/mit_fusion/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/seb/workspace/project/bevfusion_mit/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 187, in forward
    outputs = self.forward_single(
  File "/home/seb/anaconda3/envs/mit_fusion/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/seb/workspace/project/bevfusion_mit/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 269, in forward_single
    losses = head.loss(gt_bboxes_3d, gt_labels_3d, pred_dict)
  File "/home/seb/anaconda3/envs/mit_fusion/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 214, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/seb/workspace/project/bevfusion_mit/bevfusion/mmdet3d/models/heads/bbox/transfusion.py", line 607, in loss
    ) = self.get_targets(gt_bboxes_3d, gt_labels_3d, preds_dicts[0])
  File "/home/seb/workspace/project/bevfusion_mit/bevfusion/mmdet3d/models/heads/bbox/transfusion.py", line 382, in get_targets
    res_tuple = multi_apply(
  File "/home/seb/workspace/libraries/mmdetection220/mmdetection/mmdet/core/utils/misc.py", line 30, in multi_apply
    return tuple(map(list, zip(*map_results)))
  File "/home/seb/workspace/project/bevfusion_mit/bevfusion/mmdet3d/models/heads/bbox/transfusion.py", line 459, in get_targets_single
    assign_result = self.bbox_assigner.assign(
  File "/home/seb/workspace/project/bevfusion_mit/bevfusion/mmdet3d/core/bbox/assigners/hungarian_assigner.py", line 127, in assign
    matched_row_inds, matched_col_inds = linear_sum_assignment(cost)
  File "/home/seb/anaconda3/envs/mit_fusion/lib/python3.8/site-packages/scipy/optimize/_lsap.py", line 100, in linear_sum_assignment
    raise ValueError("matrix contains invalid numeric entries")
ValueError: matrix contains invalid numeric entries
--------------------------------------------------------------------------
Primary job  terminated normally, but 1 process returned
a non-zero exit code. Per user-direction, the job has been aborted.
--------------------------------------------------------------------------
--------------------------------------------------------------------------
mpirun detected that one or more processes exited with non-zero status, thus causing
the job to be terminated. The first process to do so was:

  Process name: [[5882,1],1]
  Exit code:    1
--------------------------------------------------------------------------
```
Btw I changed `workers_per_gpu` and` samples_per_gpu `to smaller number, whether it will lead to this error?
Thank you for your advice ！:D

kentang-mit, 2022-10-15:
I saw that before. I think it is somehow related to the implementation of TransFusion head, and one very possible reason is that you did not use a correct lr schedule. I'll release more details in next month.

skyezzz, 2022-10-15:
> I saw that before. I think it is somehow related to the implementation of TransFusion head, and one very possible reason is that you did not use a correct lr schedule. I'll release more details in next month.

Got that Thank you for your reply!

closed 2022-10-15 by skyezzz

xxlbigbrother, 2022-10-28:
> I saw that before. I think it is somehow related to the implementation of TransFusion head, and one very possible reason is that you did not use a correct lr schedule. I'll release more details in next month.

Hi, i meet the same problem when I train bevfusion-L. I check the settings of TranFusion's lr scedule, and I find that it is the same as settings in your repo. Can you show more details about the solution of th problem?