mit-han-lab/bevfusion #188: CUDA Memory Issue

hehualin-tut, 2022-10-15:
First of all, thanks for the open source.
Currently I have OOM issues due to the device, therefore，I thought modifying the ```samples_per_gpu``` in [default.yaml](https://github.com/mit-han-lab/bevfusion/blob/main/configs/nuscenes/default.yaml) might solve the problem, But it turned out not to be what I thought, and new problems arose：
[image]
What do I need to do if I need to reduce things like batchsize?

This is my device situation：
[image]

kentang-mit, 2022-10-15:
The way you modified batch size is correct. However, your GPU memory might not be large enough to train the model at batch size = 2. Besides, did you install mmcv 1.4.0 as suggested?

hehualin-tut, 2022-10-16:
> The way you modified batch size is correct. However, your GPU memory might not be large enough to train the model at batch size = 2. Besides, did you install mmcv 1.4.0 as suggested?

I'm sure mmcv is installed.  I'm using the Dockerfile you gave, but modified the pytorch version and cuda version due to the device, maybe it's a problem in this place? But my test stage is fine, which is very confusing. By the way, When batch size is 4, only OOM is prompted, and no new error appears.

kentang-mit, 2022-10-17:
What PyTorch version are you using? Previously I remember that our code has some problem with PyTorch >= 1.11.0. Also, there are some people reporting that CUDA 11.3 might be the best version.

hehualin-tut, 2022-10-17:
@kentang-mit 
> What PyTorch version are you using? Previously I remember that our code has some problem with PyTorch >= 1.11.0. Also, there are some people reporting that CUDA 11.3 might be the best version.
[image]

mqm-slice, 2022-11-02:
[image]
I also have this issue. My pytorch version is 1.10.1 py3.8_cuda11.3. MMCV1.4.0 has also been installed. What should i do?

mqm-slice, 2022-11-03:
> What about the mmdetection version? @mqm-slice Please ensure that you have installed 2.20.0 for mmdet. Also, please also make sure that `mmcv-full==1.4.0` is installed. The error seems to be related to the fact that you haven't compiled the corresponding CUDA extension in mmcv correctly on your device.
I reinstalled MMCV1.4.0 and the problem is now fixed. Thank you.

kentang-mit, 2022-11-03:
Sounds good! Closed since the problem is resolved.