mit-han-lab/bevfusion #194: Cuda execution failed when running visualization

carlinds, 2022-10-18:
Hi,

I'm trying to run the following vizualisation:
`torchpack dist-run -np 1 python tools/visualize.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml --checkpoint pretrained/bevfusion-det.pth --out-dir viz/fusion-det-pred --mode pred`

I'm using the provided docker image so the environment should be identical to what's described [here](https://github.com/mit-han-lab/bevfusion#prerequisites). I have also run `python setup.py develop`.

The only change I've made is to reduce the number of samples per gpu to 1 in [this config file](https://github.com/mit-han-lab/bevfusion/blob/main/configs/nuscenes/default.yaml#L261). 

However, when running the above vizualisation command I get the following error:
  
    /opt/conda/lib/python3.8/site-packages/torch/functional.py:445: UserWarning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (Triggered internally at  /opt/conda/conda-bld/pytorch_1639180588308/work/aten/src/ATen/native/TensorShape.cpp:2157.)
      return _VF.meshgrid(tensors, **kwargs)  # type: ignore[attr-defined]
    2022-10-18 13:51:26,375 - mmdet - INFO - load checkpoint from http path: https://github.com/SwinTransformer/storage/releases/download/v1.0.0/swin_tiny_patch4_window7_224.pth
    load checkpoint from local path: pretrained/bevfusion-det.pth
      0% 0/10 [00:01<?, ?it/s]
    Traceback (most recent call last):
      File "tools/visualize.py", line 173, in <module>
        main()
      File "tools/visualize.py", line 95, in main
        outputs = model(**data)
      File "/opt/conda/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
        return forward_call(*input, **kwargs)
      File "/opt/conda/lib/python3.8/site-packages/torch/nn/parallel/distributed.py", line 886, in forward
        output = self.module(*inputs[0], **kwargs[0])
      File "/opt/conda/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
        return forward_call(*input, **kwargs)
      File "/opt/conda/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 98, in new_func
        return old_func(*args, **kwargs)
      File "/home/s0001038/repos/github/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 187, in forward
        outputs = self.forward_single(
      File "/opt/conda/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 98, in new_func
        return old_func(*args, **kwargs)
      File "/home/s0001038/repos/github/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 244, in forward_single
        feature = self.extract_lidar_features(points)
      File "/home/s0001038/repos/github/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 132, in extract_lidar_features
        x = self.encoders["lidar"]["backbone"](feats, coords, batch_size, sizes=sizes)
      File "/opt/conda/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
        return forward_call(*input, **kwargs)
      File "/opt/conda/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 98, in new_func
        return old_func(*args, **kwargs)
      File "/home/s0001038/repos/github/bevfusion/mmdet3d/models/backbones/sparse_encoder.py", line 116, in forward
        x = self.conv_input(input_sp_tensor)
      File "/opt/conda/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
        return forward_call(*input, **kwargs)
      File "/home/s0001038/repos/github/bevfusion/mmdet3d/ops/spconv/modules.py", line 132, in forward
        input = module(input)
      File "/opt/conda/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
        return forward_call(*input, **kwargs)
      File "/home/s0001038/repos/github/bevfusion/mmdet3d/ops/spconv/conv.py", line 163, in forward
        outids, indice_pairs, indice_pair_num = ops.get_indice_pairs(
      File "/home/s0001038/repos/github/bevfusion/mmdet3d/ops/spconv/ops.py", line 92, in get_indice_pairs
        return get_indice_pairs_func(
    RuntimeError: mmdet3d/ops/spconv/src/indice_cuda.cu 124
    cuda execution failed with error 2
    --------------------------------------------------------------------------
    Primary job  terminated normally, but 1 process returned
    a non-zero exit code. Per user-direction, the job has been aborted.
    --------------------------------------------------------------------------
    --------------------------------------------------------------------------
    mpirun detected that one or more processes exited with non-zero status, thus causing
    the job to be terminated. The first process to do so was:
    
      Process name: [[25106,1],0]
      Exit code:    1
    --------------------------------------------------------------------------

VeeranjaneyuluToka, 2022-10-18:
Am not sure if this is of any help, i was facing same issue with training as well. But this is not all the time. when i try multiple times, some time i use to get the above error and some times it works fine. 

Thought it might be helpful though it is not a solution. Sorry if that is not the case.

kentang-mit, 2022-10-18:
May I ask what is the GPU you are using? You might be running OOM.

kentang-mit, 2022-10-18:
It's slightly more than 32G for fusion models. But results can be reproduced with <32G memory if you reduce the batch size.

closed 2022-10-24 by carlinds

yyxr75, 2023-01-02:
Hello,
I have been trained and tested this repo on nuscenes dataset and it showed no problem. So I started modifying bevfusion for my own dataset. The training step was finished perfectly and the 20th epoch trained model is saved in ./tools/run. However, when I tried to test it on my test dataset, the error popped out.

I executed `torchpack dist-run -np 1 python test.py`, the residual parse config such as model's path etc, are given default in code. I have 5 Tesla T4 GPUs with 15GB memory each, which is bigger than 12GB. Besides, I noticed that the code is error-free when training, but not in testing. The memory at the same step of code is different. Why?

>samples_per_gpu: 1
workers_per_gpu: 0

The environment of my docker container is as below: 

> absl-py                  1.3.0
addict                   2.4.0
anyio                    3.6.2
argon2-cffi              21.3.0
argon2-cffi-bindings     21.2.0
arrow                    1.2.3
asttokens                2.2.1
attrs                    22.2.0
backcall                 0.2.0
beautifulsoup4           4.11.1
bleach                   5.0.1
cachetools               5.2.0
certifi                  2022.12.7
cffi                     1.15.1
charset-normalizer       2.1.1
comm                     0.1.2
contourpy                1.0.6
cycler                   0.11.0
debugpy                  1.6.4
decorator                5.1.1
defusedxml               0.7.1
descartes                1.1.0
entrypoints              0.4
executing                1.2.0
fastjsonschema           2.16.2
fire                     0.5.0
flit_core                3.6.0
fonttools                4.38.0
fqdn                     1.5.1
google-auth              2.15.0
google-auth-oauthlib     0.4.6
grpcio                   1.51.1
h5py                     3.7.0
idna                     3.4
importlib-metadata       5.2.0
importlib-resources      5.10.1
ipykernel                6.19.4
ipython                  8.7.0
ipython-genutils         0.2.0
ipywidgets               8.0.4
isoduration              20.11.0
jedi                     0.18.2
Jinja2                   3.1.2
joblib                   1.2.0
jsonpointer              2.3
jsonschema               4.17.3
jupyter                  1.0.0
jupyter_client           7.4.8
jupyter-console          6.4.4
jupyter_core             5.1.1
jupyter-events           0.5.0
jupyter_server           2.0.5
jupyter_server_terminals 0.4.3
jupyterlab-pygments      0.2.2
jupyterlab-widgets       3.0.5
kiwisolver               1.4.4
llvmlite                 0.39.1
loguru                   0.6.0
Markdown                 3.4.1
MarkupSafe               2.1.1
[... 138 lines cut ...]
    return value.to(dtype) if is_eligible else value
RuntimeError: CUDA error: an illegal memory access was encountered
CUDA kernel errors might be asynchronously reported at some other API call,so the stacktrace below might be incorrect.
For debugging consider passing CUDA_LAUNCH_BLOCKING=1.
Aborted (core dumped)

Below is the memory difference in the same code step:

in 
> 'bevfusion/mmdet3d/models/necks/generalized_lss.py'

> 
         line 96: line 101

      .......
            laterals[i] = torch.cat([laterals[i], x], dim=1)  # 2214MB GPU test/ 3074MB GPU train
            laterals[i] = self.lateral_convs[i](laterals[i])  # 11618MB GPU test/ 3074MB GPU train
            laterals[i] = self.fpn_convs[i](laterals[i])

        # build outputs
        outs = [laterals[i] for i in range(used_backbone_levels)]  # 11618MB GPU test/3114MB GPU train

        # build outputs
        outs = [laterals[i] for i in range(used_backbone_levels)]  # 11618MB GPU test/3114MB GPU train
        return tuple(outs)