mit-han-lab/bevfusion #196: Reproduction problems on map segmentation

1349949, 2022-10-20:
Thanks for your nice work.
But i have some problems in reproducing the result of bevfusion-map-segmentation.
I samply run the following script:
_torchpack dist-run -np 8 python tools/train.py configs/nuscenes/seg/fusion-bev256d2-lss.yaml_

But i only get **map/mean/iou@max: 0.5689**  .
However, the final result you mentioned is **0.6295**
Any advice？

kentang-mit, 2022-10-20:
We haven't released the full config for fusion models. You may have a try for the single modality model and see whether the performance matches with what we provided.

1349949, 2022-10-21:
> We haven't released the full config for fusion models. You may have a try for the single modality model and see whether the performance matches with what we provided.

Yes, i add the following configs to the **fusion-bev256d2-lss.yaml**  to train the model:
**optimizer_config:
  grad_clip:
    max_norm: 35
    norm_type: 2

lr_config:
  policy: cyclic

momentum_config:
  policy: cyclic**
  
Do i need to change any other configs (such as img size, model size) to reproduce the seg results?  The result gap between my reproduction and the paper is so large.

My following work is based on the bevfusion. I'm really in great need of the config file.

kentang-mit, 2022-10-21:
Please send us an Email and I will provide you with the configs for map segmentation. Thanks.

closed 2022-10-21 by kentang-mit

jiajiaen, 2022-10-31:
I had the same problem when I reproduced bevfusion-map-segmentation.
I loaded the swint-unimages-pretrained.pth and lidar-only-seg.pth weights to initialize the model, and added the following configuration information to fusion-bev256d2-lss.yaml:
              optimizer_config:
                    grad_clip:
                    max_norm: 35
                    norm_type: 2

              lr_config:
                   policy: cyclic

             momentum_config:
                   policy: cyclic

I run the following script: torchpack dist-run -np 8 python tools/train.py configs/nuscenes/seg/fusion-bev256d2-lss.yaml
The trained map/NDS is similar to the above results. My following work is also based on bevfusion, and I really hope to get the config file.

best,
Jiapeng