mit-han-lab/bevfusion #200: problem with debugging on pycharm

Z-chocking, 2022-10-24:
I wanted to debug the project on pycharm and removed the `dist.init()`  function in reference to[ what others had done](https://github.com/mit-han-lab/bevfusion/issues/143)。
Then I run it in pycharm debug mode:
`tools/train.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml --run-dir nbev_result/`
But this is an error：
[image]
What should I do？It seems that I didn't initialize the distribution

kentang-mit, 2022-10-25:
Sorry but I'm not an expert in Pycharm. Would you mind having a look at this issue: https://github.com/mit-han-lab/bevfusion/issues/172 to see whether it is possible to replace torchpack.distributed with torch.distributed? This might be helpful. If it won't work, I would suggest you to use the IDE to edit the code and debugging using command line tools.

Z-chocking, 2022-10-25:
Thanks for your reply.I have solved this problem by modifying some code in this work

closed 2022-10-25 by Z-chocking