mit-han-lab/bevfusion #202: problems caused by spconv

MakiNaruto, 2022-10-26:
Thanks for your great work, but i get some problem that i can't solve.
my environment is:
3090, cuda11.3. use docker in README.md

I run the code ```torchpack dist-run -np 1 python tools/train.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml```     it's fine
torchpack dist-run -np 1 python configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/default.yaml      it's bad
"
I try the method in #148, and get new error like this:
```
Traceback (most recent call last):
  File "tools/train.py", line 87, in <module>
    main()
  File "tools/train.py", line 76, in main
    train_model(
  File "/home/bevfusion/bevfusion-main/mmdet3d/apis/train.py", line 126, in train_model
    runner.run(data_loaders, [("train", 1)])
  File "/opt/conda/lib/python3.8/site-packages/mmcv/runner/epoch_based_runner.py", line 127, in run
    epoch_runner(data_loaders[i], **kwargs)
  File "/home/bevfusion/bevfusion-main/mmdet3d/runner/epoch_based_runner.py", line 14, in train
    super().train(data_loader, **kwargs)
  File "/opt/conda/lib/python3.8/site-packages/mmcv/runner/epoch_based_runner.py", line 50, in train
    self.run_iter(data_batch, train_mode=True, **kwargs)
  File "/opt/conda/lib/python3.8/site-packages/mmcv/runner/epoch_based_runner.py", line 29, in run_iter
    outputs = self.model.train_step(data_batch, self.optimizer,
  File "/opt/conda/lib/python3.8/site-packages/mmcv/parallel/distributed.py", line 52, in train_step
    output = self.module.train_step(*inputs[0], **kwargs[0])
  File "/home/bevfusion/bevfusion-main/mmdet3d/models/fusion_models/base.py", line 78, in train_step
    losses = self(**data)
  File "/opt/conda/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/opt/conda/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/bevfusion/bevfusion-main/mmdet3d/models/fusion_models/bevfusion.py", line 187, in forward
    outputs = self.forward_single(
  File "/opt/conda/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/bevfusion/bevfusion-main/mmdet3d/models/fusion_models/bevfusion.py", line 256, in forward_single
    assert len(features) == 1, features
AssertionError: [tensor([[[[0.0000, 0.0000, 0.0000,  ..., 0.0000, 0.0000, 0.0000],
          [0.0000, 0.0000, 0.0000,  ..., 0.0000, 0.0000, 0.0000],
          [0.0000, 0.0000, 0.0000,  ..., 0.0000, 0.0000, 0.0000],
          ......
```

MakiNaruto, 2022-10-27:
I founded my error。#51

closed 2022-10-27 by MakiNaruto

JiSuanJiDaWang, 2022-11-18:
> I founded my error。#51

Can you please describe more specifically how you solve your problem? Thanks.